ARMmbed/mbed-os #8833: Bug: Trying to compile with bootloader for non-bootloader target causes python error

screamerbg, 2018-11-21:
### Description

Python error occurs while trying to compile with a bootloader for a target that doesn't have `bootloader_supported` flag.

```
[mbed] Working path "D:\Work\smcc-tests" (program)
[mbed] Detected "DISCO_F746NG" connected to "E:" and using com port "COM5"
Building library mbed-build (DISCO_F746NG, ARM)
Scan: smcc-tests
Bootloader not supported on this target.
Failed to build library
Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|
Traceback (most recent call last):
  File "D:\Work\smcc-tests\mbed-os\tools\test.py", line 288, in <module>
    status = print_report_exporter.report(build_report)
  File "D:\Work\smcc-tests\mbed-os\tools\test_exporters.py", line 92, in report
    return self.exporter_print(test_summary_ext, print_log_for_failures=print_log_for_failures)
  File "D:\Work\smcc-tests\mbed-os\tools\test_exporters.py", line 345, in exporter_print
    raise Exception("'test_run' did not have a 'result' value")
Exception: 'test_run' did not have a 'result' value
[ERROR] 'test_run' did not have a 'result' value
```

CC @cmonr @theotherjimmy 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

theotherjimmy, 2018-11-26:
@screamerbg What do you expect to happen?

screamerbg, 2019-03-28:
This is fixed in the recent version of Mbed OS.

closed 2019-03-28 by screamerbg