ARMmbed/mbed-os #8842: Add better errors in FlashIAP.cpp

janjongboom, 2018-11-22:
FlashIAP returns -1 when it sees an error, which is a bit unhelpful when a lot of things go on. E.g. when you try to mount LittleFS to a FlashIAPBlockDevice whose addresses are read-sector aligned, but not erase-sector aligned the blockdevice init goes through fine, but formatting (after mounting failed) fails with `-1` ([source](https://github.com/ARMmbed/mbed-os/blob/60b5547b652bbb840afc6e0212d4443890aaf657/drivers/FlashIAP.cpp#L166)). That's not helpful.

Could we add some better error codes, e.g. FLASH_NON_SECTOR_ALIGNED_ERASE with a unique error code so it's easier to found?

### Issue request type

    [ ] Question
    [X Enhancement
    [ ] Bug

janjongboom, 2019-11-01:
@adbridge In my opinion this is a defect, not an enhancement.

adbridge, 2019-11-04:
@janjongboom thanks for the update I'll update it accordingly