ARMmbed/mbed-os #8857: Silabs EFM32 watchdog timeout values

fkjagodzinski, 2018-11-23:
### Description

@scartmell-arm @donatieng I checked our watchdog tests locally and found that some of them fail on EFM32GG_STK3700. Here is the test summary:
```
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+---------+--------------------+
| target                  | platform_name   | test suite                        | test case                              | passed | failed | result  | elapsed_time (sec) |
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+---------+--------------------+
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-reset_reason   | tests-mbed_drivers-reset_reason        | 1      | 0      | OK      | 32.72              |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Restart multiple times                 | 0      | 1      | FAIL    | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, 0 ms                            | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, 500 ms                          | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, max_timeout                     | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, max_timeout exceeded            | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Stop                                   | 1      | 0      | OK      | 0.52               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | max_timeout is valid                   | 1      | 0      | OK      | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Kicking the watchdog prevents reset    | 1      | 0      | OK      | 0.03               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog reset                         | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog reset in sleep mode           | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog started again                 | 1      | 0      | OK      | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-reset_reason       | tests-mbed_hal-reset_reason            | 1      | 0      | OK      | 32.56              |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Init, 500 ms                           | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Init, max_timeout                      | 0      | 0      | SKIPPED | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Platform feature max_timeout is valid  | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Stopped watchdog can be started again  | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Update config with multiple init calls | 0      | 1      | FAIL    | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Watchdog can be stopped                | 1      | 0      | OK      | 0.52               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Kicking the watchdog prevents reset    | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog reset                         | 1      | 0      | OK      | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog reset in sleep mode           | 1      | 0      | OK      | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog started again                 | 1      | 0      | OK      | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 1000 ms                        | 1      | 0      | OK      | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 200 ms                         | 1      | 0      | OK      | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 3000 ms                        | 0      | 1      | FAIL    | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 500 ms                         | 1      | 0      | OK      | 0.0                |
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+---------+--------------------+
```
This is because EFM32 targets have a list of predefined watchdog timeouts that may be set.

https://github.com/ARMmbed/mbed-os/blob/7995e8beacd87a44f1cf4bd873316e8cfca7cc19/targets/TARGET_Silicon_Labs/TARGET_EFM32/emlib/inc/em_wdog.h#L66-L84

Assuming we use the default `wdogClkSelULFRCO` 1 kHz clock, an arbitrary value like 500 ms cannot be set (the closest we can get is 513 ms in this case).

So the question is: Should we provide some additional info to the user? Or maybe explicitly state that the timeout value actually set in the peripheral may be far from the parameter value used in `hal_watchdog_init()` call?

We could also point to the `hal_watchdog_get_reload_value()` fun, that can be used to verify the timeout setting.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2018-11-23:
@ARMmbed/team-silabs Please review

fkjagodzinski, 2019-02-25:
@c1728p9, @rajkan01, I'm curious what's your opinion on this. Would a warning note in the docs be enough in this case?

c1728p9, 2019-02-26:
The [HAL spec for watchdog](https://os.mbed.com/docs/mbed-os/v5.11/porting/watchdog-port.html) states:
```
The timeout must be accurate to the nearest millisecond.
```

This seems a bit strict. Do any of our use cases this much accuracy? Maybe this requirement could be relaxed to something like +-50% of the target value? @scartmell-arm what are your thoughts?

donatieng, 2019-02-27:
Requiring one millisecond accuracy is unrealistic on most targets as oscillators used for watchdog peripherals are often imprecise (and they don't need to be).

I think we should change this requirement to something like:
* The watchdog should trigger at or after the timeout value
* The watchdog should trigger before twice the timeout value

@c1728p9 @fkjagodzinski @rajkan01 @scartmell-arm thoughts?

c1728p9, 2019-02-27:
Sounds good to me.

stevew817, 2019-03-08:
From the ARMmbed/team-silabs perspective I support the proposal from @donatieng

stevew817, 2019-04-25:
@c1728p9 @donatieng Was this issue followed up on? Can we drive towards closure?

donatieng, 2019-04-25:
Apologies for the late reply, I'm following up with @rajkan01 to make sure this is reflected within #10066

rajkan01, 2019-04-25:
Needs to specify watchdog timeout value in targets.json file based on target on the new implementation 
"hw-watchdog_timeout": {
                "help": "Define the timeout in ms value LowPowerTicker to do HW kick",
                "value": "800",
                "macro_name": "HW_WATCHDOG_TIMEOUT"
            }
Here value 800/2 configured as hardware watchdog timeout,i will update watchdog.md file once the PR#10066 merged into feature-watchdog branch.

@fkjagodzinski I thing we need to configure this value before running watchdog test in the hardware...Please check

fkjagodzinski, 2019-04-26:
> @fkjagodzinski I thing we need to configure this value before running watchdog test in the hardware...Please check

@rajkan01 Yes, the `HW_WATCHDOG_TIMEOUT` parameter is used in tests that come with the new implementation of the driver/platform layer.

https://github.com/ARMmbed/mbed-os/blob/fdc1be578f0c2e77b485282dc30cfe4bc8181243/TESTS/mbed_platform/watchdog_mgr_reset/main.cpp#L90

fkjagodzinski, 2019-04-26:
When I'm back in the office, I'll update all affected tests (watchdog & reset_reason) to match new timing requirements.
> * The watchdog should trigger at or after the timeout value
> * The watchdog should trigger before twice the timeout value

fkjagodzinski, 2019-05-15:
The changes for tests are ready. Some minor updates will be needed after #10066 is merged, so I'll open a PR as soon as #10066 is merged.

fkjagodzinski, 2019-07-08:
All correct on current `master` (492d10c).
```
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+--------+--------------------+
| target                  | platform_name   | test suite                        | test case                              | passed | failed | result | elapsed_time (sec) |
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+--------+--------------------+
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Restart multiple times                 | 1      | 0      | OK     | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, 100 ms                          | 1      | 0      | OK     | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Start, max_timeout                     | 1      | 0      | OK     | 0.03               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | Stop                                   | 1      | 0      | OK     | 0.22               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog       | max_timeout is valid                   | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Kicking the Watchdog prevents reset    | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog reset                         | 1      | 0      | OK     | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog reset in sleep mode           | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_drivers-watchdog_reset | Watchdog started again                 | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Init, 100 ms                           | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Init, max_timeout                      | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Platform feature max_timeout is valid  | 1      | 0      | OK     | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Stopped watchdog can be started again  | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Update config with multiple init calls | 1      | 0      | OK     | 0.02               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog           | Watchdog can be stopped                | 1      | 0      | OK     | 0.21               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Kicking the Watchdog prevents reset    | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog reset                         | 1      | 0      | OK     | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog reset in sleep mode           | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_reset     | Watchdog started again                 | 1      | 0      | OK     | 0.01               |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 1000 ms                        | 1      | 0      | OK     | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 200 ms                         | 1      | 0      | OK     | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 3000 ms                        | 1      | 0      | OK     | 0.0                |
| EFM32GG_STK3700-GCC_ARM | EFM32GG_STK3700 | tests-mbed_hal-watchdog_timing    | Timing, 500 ms                         | 1      | 0      | OK     | 0.0                |
+-------------------------+-----------------+-----------------------------------+----------------------------------------+--------+--------+--------+--------------------+

```

PR #10839 (2019-06-26): Update watchdog timing requirements
### Description

Relax the timing requirements for watchdog according to https://github.com/ARMmbed/mbed-os/issues/8857#issuecomment-467838213:
* The watchdog should trigger at or after the timeout value.
* The watchdog should trigger before twice the timeout value.

Fixes #8857 

Docs update in https://github.com/ARMmbed/mbed-os-5-docs/pull/1088

~**NOTE:** Marking this as a draft -- we should wait until #10777 is merged.~

### Pull request type

    [ ] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [x] Test [... cut]
closed 2019-07-08 by fkjagodzinski