ARMmbed/mbed-os #8858: Set default baudrate and pin values for default printf

masoudr, 2018-11-23:
### Description
As you might know, there is a serial that associated with `printf` function. I was wondering where the related serial pins and baud rate set, so I can change the pins and set the default baud rate.
Currently, I'm using this method to redefine `printf` but It would be easier to change it on source files.

```
#include "mbed.h"

#undef printf
#define printf(a, ...) pc.printf(a, ##__VA_ARGS__)

Serial pc(PA_9, PA_10); // tx, rx

int main(){
    pc.baud(115200);
    printf("Hello! \n");
}

```
### Issue request type
    [x] Question
    [ ] Enhancement
    [ ] Bug

mstfkaratas, 2018-11-25:
You dont need to change source code. You can use following configuration options in mbed_app.json:

```
"platform.stdio-baud-rate": 115200,
"target.stdio_uart_tx": "PA_9",
"target.stdio_uart_rx": "PA_10",
```

j3hill, 2018-11-30:
In order to use `readable`, your serial port should be associated with a stream. This can be done in the constructor for the Serial class. For example: 
`Serial pc(PA_9, PA_10, <name of stream>, <OPTIONAL baudrate>); `
Here is more information: https://os.mbed.com/docs/v5.10/mbed-os-api-doxy/classmbed_1_1_serial.html

masoudr, 2018-12-01:
@j3hill Thanks, but I thought that I can override this in the `mbed_app.json` file. Anyway, I'm closing this issue as already answered.

closed 2018-12-01 by masoudr