ARMmbed/mbed-os #8864: TRNG on L4 pulls rug out from USB

cblack-pointlabs, 2018-11-25:
### Description

**Target:** STM32L4x
**Scope:** USB and TRNG co-existence

In system_clock.c, the USB/SDMMC/TRNG source clock is set to either MSI or PLLSAI1, depending on the clock source (HSE and HSI use PLLSAI1 for 48MHz source, and MSI use MSI for 48MHz source).

trng_api.c does not take this into account, and incorrectly sets the 48MHz clock source, in contradiction to what it was set to in system_clock.c

#### Bug Reproduction
1. In json config, choose USE_PLL_MSI for clock source
2. Use trng functionality (such as mbedtls_hardware_poll function call)
3. USB operation ceases as soon as trng_init is called, as the 48MHz clock source is incorrectly changed to RCC_RNGCLKSOURCE_PLL, invalidating how it was set up in system_clock.c

Recommend a change in trng_api.c to choose the correct 48MHz source as it was set in system_clock.c, or remove this code as the 48MHz clock source was already set in system_clock.c, correct?

```
#if ((CLOCK_SOURCE) & USE_PLL_MSI)
    PeriphClkInitStruct.RngClockSelection = RCC_RNGCLKSOURCE_MSI;
#else
    PeriphClkInitStruct.RngClockSelection = RCC_RNGCLKSOURCE_PLLSAI1;
#endif
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cblack-pointlabs, 2018-11-25:
Specific targets system_clock.c:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L475xG/TARGET_DISCO_L475VG_IOT01A/system_clock.c

bcostm, 2018-11-26:
Hi
Thanks for this analysis and reporting

I also agree to remove the TRNG clock setting in the trng_api.c file and add it in the system_clock.c file instead like we do for the USB. Just maybe add a "#ifdef DEVICE_TRNG". I will send a PR so you can test it and comment.