ARMmbed/mbed-os #8865: SecureStore requires MBEDTLS_CIPHER_MODE_CTR to compile

artokin, 2018-11-26:
### Description

SecureStore requires mbedtls configuration flag MBEDTLS_CIPHER_MODE_CTR in order to compile.

We are using custom mbedtls configuration but we don't use SecureStore and we get following compilation error if MBEDTLS_CIPHER_MODE_CTR is not defined:

_22:54:41 Compile [ 51.3%]: SecureStore.cpp
22:54:41 
[Error] SecureStore.cpp@109,0:  #20: identifier "mbedtls_aes_crypt_ctr" is undefined
22:54:42 [ERROR] "./mbed-os/features/storage/kvstore/securestore/SecureStore.cpp", line 109: Error:  #20: identifier "mbedtls_aes_crypt_ctr" is undefined
22:54:42 ./mbed-os/features/storage/kvstore/securestore/SecureStore.cpp: 0 warnings, 1 error
22:54:42 
22:54:42 [mbed] ERROR: "/usr/bin/python" returned error._

Would it be possible to have conditional compilation in the SecureStore component in order to avoid such compilation error? If all prerequisites are not met then component should not compile. 

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

0xc0170, 2018-11-26:
@ARMmbed/mbed-os-storage Please review

davidsaada, 2018-11-26:
Will push a fix soon to fix this - SecureStore will depend on this flag (as well as on the CMAC one), meaning that by default, SecureStore will be enabled, but if a custom MBETLS config file is there - SecureStore will only be enabled if these flags are included.

Patater, 2018-11-28:
K64F has PSA enabled by default. PSA depends on the SecureStore for implementing PSA internal storage (ITS). Will we also be making TARGET_PSA get conditionally defined based on whether SecureStore is available?

CC @alzix

alzix, 2018-11-29:
@Patater it is not related to PSA internal storage PSA internal storage depends on internal TDBStore.
Thus i think it is not related to `TARGET_PSA`

closed 2018-12-11 by 0xc0170