ARMmbed/mbed-os #8870: Feature: QSPI support on DISCO_F769NI

screamerbg, 2018-11-26:
### Description

The DISCO_F769NI has quite large QSPI on board, but unfortunately the target support is lacking the QSPI "device_has" flag and also qspi_s definies in objects.h. Could you please add QSPI (and QSPIF feature) support for it?

CC @ARMmbed/team-st-mcd 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

screamerbg, 2018-11-26:
Also cc @MarceloSalazar

jeromecoutant, 2018-11-27:
Hi
Not so easy as DISCO_F769NI  is using MX25L51245G which is not described yet in the test directory.
Thx

jeromecoutant, 2018-11-27:
Maybe you can check 
https://github.com/jeromecoutant/mbed/commits/DEV_QSPI_TEST
Thx

jeromecoutant, 2018-11-29:
Current status with the above branch:

mbed test -t xxx -m xxx -v –n components-storage-blockdevice-component_qspif-tests-block_device-qspif

- F469: OK
- F746: OK
- F769: FAILED (wrong device Id)
- L475: OK
- L476: OK
- L496: FAILED (hard fault)

@screamerbg could you add QSPI expert in this issue discussion ?

Thx

screamerbg, 2018-11-29:
CC @offirko

offirko, 2018-11-29:
@jeromecoutant - could you attach the full logs at the Jira or here , or send me by mail?
I do not have the F769 or L496 boards, and I cant find them at any of the CI sites either.

CC @maciejbocianski

dannybenor, 2018-11-29:
@jeromecoutant is F769 passing the HAL tests for QSPI?

jeromecoutant, 2018-11-29:
No, but DISCO_F769 has a new QSPI : MX25L51245G
which doesn't have a clean flash config configuration file...

offirko, 2018-11-29:
@jeromecoutant - I believe qspif-blockdevice tests should only be run on the new targets after tests-mbed_hal-qspi tests are successful.

offirko, 2018-12-10:
@screamerbg - I've received the DISCO_F769NI you sent. After making a small update to target hal and QSPIF BD fix the device works well. I'm not sure whether I can use this PR to deliver the changes.. or should I open a new PR?

screamerbg, 2018-12-10:
@offirko Thanks for your work to enable this board. I think that it would be great if its part of this PR if @jeromecoutant doesn't object.

offirko, 2018-12-10:
Note that after my changes QSPIFBlockDevice tests pass successfully, but 'tests-mbed_hal-qspi' still Fail.
This is something that I can not fix - I think it will require @jeromecoutant or @maciejbocianski to add specific setup for this board to the test.

maciejbocianski, 2018-12-11:
This target is not supported in mbed_hal-qspi test yet, I will add next week at the latest

offirko, 2018-12-11:
Added PR #9057 which enables QSPIFBlockDevice to work properly with this board (once hal qspi test are supported)
https://github.com/ARMmbed/mbed-os/pull/9057

jeromecoutant, 2018-12-12:
Hi
Status of 
https://github.com/jeromecoutant/mbed/commits/DEV_QSPI_TEST
+ #9057:

| target           | platform_name | test suite                                                              | result | elapsed_time (sec) | copy_method |
|------------------|---------------|-------------------------------------------------------------------------|--------|--------------------|-------------|
| DISCO_F769NI-ARM | DISCO_F769NI  | components-storage-blockdevice-component_qspif-tests-block_device-qspif | OK     | 21.67              | default     |
| DISCO_F769NI-ARM | DISCO_F769NI  | tests-mbed_hal-qspi                                                     | FAIL   | 18.73              | default     |

PR #9089 (2018-12-14): STM32 QSPI update
### Description

Few patches around QSPI feature within STM32:

- DISCO_F746NG is now enabled
- DISCO_F769NI is now enabled
- DISCO_F469NI is now enabled
- DISCO_L496AG is now enabled
- QSPI pins have been removed for boards that don't have embeded QSPI
- dual bank QSPI pins have been removed
- frequency setting is corrected

QSPIF component tests and QSPI feature tests have been executed.

This fixes #8870 

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
closed 2018-12-14 by cmonr