ARMmbed/mbed-os #8892: These radio resources no longer work on microbit

rhubarbdog, 2018-11-28:
###Description

These resource links for radio examples no longer work on microbit
Using the Microbit's TX/RX radio for communication between boards: <a href="https://os.mbed.com/teams/microbit/code/microbit-simple-radio-rx/">radio-rx</a> & <a href="https://os.mbed.com/teams/microbit/code/microbit-simple-radio-tx/">radio-tx</a>.
I'm using the <a href="https://ide.mbed.com/compiler/#nav:/;">online compiler</a> 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

pan-, 2018-11-28:
Could you clarify what does no longer work ?

rhubarbdog, 2018-11-28:
The links radio-tx and radio-rx point to arm mbed microbit c++ resources.
I have imported them into my online compiler
Compiled, saved and flashed. Tx onto 1 microbit and Rx onto another. When i press a button on Tx i expect it to broadcast a datagram, which should be received by Rx. Nothing happens
The radio works on these microbits i can show similar code running in micropython.

pan-, 2018-11-28:
Did you update the library when you imported them in the compiler ?

adbridge, 2021-06-07:
As there has been no update on this for over 2 years I am closing it. Please re-open if this is still a valid issue.

closed 2021-06-07 by adbridge