ARMmbed/mbed-os #8894: EventFlags wait timeout units

sarahmarshy, 2018-11-28:
### Description

The EventFlags class does not specify a unit of time for its timeout. 
* [wait_all](https://github.com/ARMmbed/mbed-os/blob/de6ba91644c4b2396d9445ada1554b2487652a24/rtos/EventFlags.h#L90)
* [wait_any](https://github.com/ARMmbed/mbed-os/blob/de6ba91644c4b2396d9445ada1554b2487652a24/rtos/EventFlags.h#L100)

I tried to follow the callstack, but I didn't really get anywhere with the units for this timeout.

### Issue request type
    [ ] Question
    [ X] Enhancement
    [ ] Bug

kjbracey-arm, 2018-11-29:
Possible that this carries through a lot of stuff - many descriptions in the `rtos` namespace including that are cut-and-paste from CMSIS-RTOS documentation following that form.

Basic rule is that every time specified in the `rtos` namespace is in CMSIS-RTOS "ticks", which CMSIS-RTOS never defines itself. But for Mbed OS we have specified that the RTOS tick is a millisecond (see `Kernel::get_ms_count()`).

I see some things like `ThisThread::flags_wait_all_for` have the same wording, but changed the variable name to `millisec`, which seems like it might do?

Also, the fact that it's "ticks" is also significant - waiting for 5 ticks will take 4.000 to 5.000 milliseconds, depending on phase relative to the next tick. It doesn't aim for precisely 5 millisecond duration (unlike `wait_us(5000)`.

loverdeg-ep, 2019-02-12:
#9077

kegilbert, 2019-02-22:
https://github.com/ARMmbed/mbed-os/pull/9670

kegilbert, 2019-02-22:
@sarahmarshy https://github.com/ARMmbed/mbed-os/pull/9670 

The above PR should have addressed this, let me know if you have remaining concerns for this issue! Otherwise it should be good to close.

closed 2019-02-25 by sarahmarshy