ARMmbed/mbed-os #8903: Realtek_RTL8195AM is printing a lot of traces

KariHaapalehto, 2018-11-29:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2018-11-29:
@KariHaapalehto CAn you add a snippet from terminal what is being printed?

cc @ARMmbed/team-realtek Please review

KariHaapalehto, 2018-11-30:
[realtek.log](https://github.com/ARMmbed/mbed-os/files/2632308/realtek.log)

Log file from start-up with realtek using mbed-os-example-wifi

0xc0170, 2018-11-30:
@KariHaapalehto Why was this closed?

KariHaapalehto, 2018-11-30:
This was accidently closed while adding log to jira ticket. Reopening

AdamZhang0124, 2018-12-27:
Hi @KariHaapalehto, 
The problem that you spotted has been solved with the pull request #9202.
The test result is attached for your information.
Thanks!
[test_log.txt](https://github.com/ARMmbed/mbed-os/files/2711601/test_log.txt)

AdamZhang0124, 2019-01-04:
Hi @KariHaapalehto ,
The solution #9202 has been merged.
Please help to check if the solution is also working from your side, then this issue can be closed.
Thanks!

samchuarm, 2019-01-04:
Hi @KariHaapalehto , can you please review it and close the ticket if the reported issue is resolved? Thanks.

KariHaapalehto, 2019-01-04:
Ticket has been resolved

closed 2019-01-04 by KariHaapalehto