ARMmbed/mbed-os #8906: LowPowerTimeout Broken

klaas019, 2018-11-29:
### Description

LowPowerTimeout is broken in mbed-os-5.10.3 and 5.10.4. The LowPowerTimeout will not work properly if the time to fire is set to 3 seconds or greater. If one of the timeouts is less than 3 seconds then all of the timeouts will fire correctly.

Test1:

Code:

#include "mbed.h"
#define NUM_TIMEOUTS 5

int main() {
    Timer timers[NUM_TIMEOUTS];
    LowPowerTimeout timeouts[NUM_TIMEOUTS];

    for (int i = 0; i < NUM_TIMEOUTS; i++) {
        timers[i].start();
        timeouts[i].attach_us(mbed::callback(&timers[i], &Timer::stop), 3000000);
    }

    wait(6); //wait for all timeouts to fire

    for (size_t i = 0; i < NUM_TIMEOUTS; i++) {
        printf("%d : %d \n\r", i, timers[i].read_us());
    }

    return 0;
}

Output:

0 : 6000538 
1 : 6013045 
2 : 6027559 
3 : 6042073 
4 : 6056589

None of the timeouts fired.

Test 2:

Code:

#include "mbed.h"
#define NUM_TIMEOUTS 5

int main() {
    Timer timers[NUM_TIMEOUTS];
    LowPowerTimeout timeouts[NUM_TIMEOUTS];

    timers[NUM_TIMEOUTS-1].start();
    timeouts[NUM_TIMEOUTS-1].attach_us(mbed::callback(&timers[NUM_TIMEOUTS-1], &Timer::stop), 2000000);

    for (int i = 0; i < NUM_TIMEOUTS - 1; i++) {
        timers[i].start();
        timeouts[i].attach_us(mbed::callback(&timers[i], &Timer::stop), 3000000);
    }

    wait(6); //wait for all timeouts to fire

    for (size_t i = 0; i < NUM_TIMEOUTS; i++) {
        printf("%d : %d \n\r", i, timers[i].read_us());
    }

    return 0;
}

Output:

0 : 3000504 
1 : 3000467 
2 : 3000431 
3 : 3000399 
4 : 2000352

All fired correctly with the last one firing at 2 seconds instead of 3. Somehow this affects the behavior of the other timeouts. 

Even setting one timeout to >= 3 seconds, the timeout will fail by itself. The number of timeouts set doesn't appear to affect this.

Tested with mbed-os-5.10.3, GCC_ARM v 6 q2, XDOT_L151CC, MTS_MDOT_F411RE, NUCLEO_F411RE.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2018-11-29:
Hi

I don't see any big change between 5.10.3 and 5.10.4 ...

About tests status, I didn't see any regression for NUCLEO_F411RE-GCC_ARM betwen all version 5.10.x

ex:
- tests-mbed_drivers-lp_timeout: OK
- tests-mbed_drivers-lp_ticker: OK

jeromecoutant, 2018-11-29:
I think my first answer may be wrong...
Please follow #8777 
You can even try the proposed patch.

klaas019, 2018-11-29:
My wording may have been confusing, the above code was working properly in 5.10.2 and stopped working in 5.10.3.

klaas019, 2018-11-29:
I tried the patch from #8777, and it didn't work. Also tried doing a git revert on the commit 023603faa7c7579d0e8ee30459fe7b55e3c76335, this also did not work.

kjbracey-arm, 2018-11-30:
If the failure is that clear-cut, then I suggest you go ahead and use `git bisect` to isolate the breaking commit. Probably the fastest way to isolate it.

klaas019, 2018-11-30:
This commit is causing the issue : a9552232e44774bf59a07db8b45f51852a07ff8c 

I have only tested with XDOT_L151CC and MTS_MDOT_F411RE, I assume this would affect other STM32 devices.

jeromecoutant, 2018-11-30:
I don't think we can revert this patch, as it is a real coding issue...

It means that this coding issue was hiding your issue...

fkjagodzinski, 2018-11-30:
> This commit is causing the issue : [a955223](https://github.com/ARMmbed/mbed-os/commit/a9552232e44774bf59a07db8b45f51852a07ff8c)
> 
> I have only tested with XDOT_L151CC and MTS_MDOT_F411RE, I assume this would affect other STM32 devices.

I can confirm this is also true for `NUCLEO_F401RE`.

fkjagodzinski, 2018-11-30:
`NUCLEO_F401RE` fails `5 s delay accuracy (attach)` test case from `tests-mbed_drivers-lp_timeout` for both master and a116d59da1c0f1393d7cfdc6d0a20710f918da64.

```
+-----------------------+---------------+-------------------------------+---------------------------------------------------+--------+--------+---------+--------------------+
| target                | platform_name | test suite                    | test case                                         | passed | failed | result  | elapsed_time (sec) |
+-----------------------+---------------+-------------------------------+---------------------------------------------------+--------+--------+---------+--------------------+
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay accuracy (attach)                       | 1      | 0      | OK      | 1.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay accuracy (attach_us)                    | 1      | 0      | OK      | 1.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay during deepsleep (attach)               | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay during deepsleep (attach_us)            | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay during sleep (attach)                   | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 1 s delay during sleep (attach_us)                | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 10 ms delay accuracy (attach)                     | 1      | 0      | OK      | 0.06               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 10 ms delay accuracy (attach_us)                  | 1      | 0      | OK      | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 5 s delay accuracy (attach)                       | 0      | 0      | ERROR   | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | 5 s delay accuracy (attach_us)                    | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback called once (attach)                     | 1      | 0      | OK      | 0.07               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback called once (attach_us)                  | 1      | 0      | OK      | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback not called when cancelled (attach)       | 1      | 0      | OK      | 0.1                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback not called when cancelled (attach_us)    | 1      | 0      | OK      | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback override (attach)                        | 1      | 0      | OK      | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Callback override (attach_us)                     | 1      | 0      | OK      | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Multiple timeouts running in parallel (attach)    | 1      | 0      | OK      | 0.08               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Multiple timeouts running in parallel (attach_us) | 1      | 0      | OK      | 0.09               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Timing drift (attach)                             | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Timing drift (attach_us)                          | 0      | 0      | SKIPPED | 0.0                |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Zero delay (attach)                               | 1      | 0      | OK      | 0.05               |
| NUCLEO_F401RE-GCC_ARM | NUCLEO_F401RE | tests-mbed_drivers-lp_timeout | Zero delay (attach_us)                            | 1      | 0      | OK      | 0.05               |
+-----------------------+---------------+-------------------------------+---------------------------------------------------+--------+--------+---------+--------------------+
```

klaas019, 2018-11-30:
Also in the same file these lines:

243: LPTICKER_RTC_time = (timeStruct.Seconds + timeStruct.Minutes * 60 + timeStruct.Hours * 60 * 60) * PREDIV_S_VALUE;

334: LPTICKER_counter += (RTC_time_tick + 24 * 60 * 60 * PREDIV_S_VALUE - LPTICKER_RTC_time);

Has similar issues now as well.

jeromecoutant, 2018-12-04:
@klaas019 could you check with #8777 + #8956 ?

klaas019, 2018-12-04:
#8777 + #8956 appears to have fixed the problem. Could you verify by running the lowpower mbed tests on other st boards?

klaas019, 2019-01-03:
Closing this issue, fixed with 5.11.1.

closed 2019-01-03 by klaas019