ARMmbed/mbed-os #8913: Build of NUCLEO_F030R8 fails when DEVICE_SERIAL=0

deece, 2018-11-30:
### Description
I am trying to build a cutdown mbed to use as part of a bootloader. In order to make the final binary as small as possible, I have historically passed: -D DEVICE_SERIAL=0 -D DEVICE_STDIO_MESSAGES=0 to the build script.

When passing these defines to tag`mbed-os-5.9.7` (I can't build master due to #8912), I get the following failure:

```
$ MBED_GCC_ARM_PATH=/usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin python build.py -c -v -m NUCLEO_F030R8 -t GCC_ARM --profile release -D DEVICE_SERIAL=0 -D USE_PLL_HSE_EXTC=0 -D USE_PLL_HSE_XTAL=0 -D DEVICE_STDIO_MESSAGES=0
...
Compile [ 28.8%]: RawSerial.cpp
Compile: /usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -g1 -DMBED_RTOS_SINGLE_THREAD -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DCMSIS_VECTAB_VIRTUAL -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DTARGET_NUCLEO_F030R8 -D__CMSIS_RTOS -DUSE_PLL_HSE_EXTC=0 -DTOOLCHAIN_GCC -DDEVICE_CRC=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M0 -DDEVICE_SPI_ASYNCH=1 -DMBED_BUILD_TIMESTAMP=1543552042.53 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_STDIO_MESSAGES=0 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL=1 -DDEVICE_USTICKER=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_SERIAL=0 -DUSE_PLL_HSE_XTAL=0 -DTARGET_STM32F030R8 -DTARGET_LIKE_CORTEX_M0 -DDEVICE_SERIAL_FC=1 @/home/adsilva/src/mbed-os/BUILD/mbed/.temp/TARGET_NUCLEO_F030R8/TOOLCHAIN_GCC_ARM/.includes_60b08cff41dbb009a5ee08b83a809415.txt -include /home/adsilva/src/mbed-os/BUILD/mbed/.temp/TARGET_NUCLEO_F030R8/TOOLCHAIN_GCC_ARM/mbed_config.h -MD -MF ../BUILD/mbed/.temp/TARGET_NUCLEO_F030R8/TOOLCHAIN_GCC_ARM/RawSerial.d -o ../BUILD/mbed/.temp/TARGET_NUCLEO_F030R8/TOOLCHAIN_GCC_ARM/RawSerial.o /home/adsilva/src/mbed-os/drivers/RawSerial.cpp
[Warning] <command-line>@0,0: "DEVICE_STDIO_MESSAGES" redefined
[Warning] <command-line>@0,0: "DEVICE_SERIAL" redefined
[Error] SerialBase.h@166,43: 'SerialIrq' has not been declared
[Error] SerialBase.h@259,5: 'serial_t' does not name a type; did you mean 'serial_s'?
[DEBUG] Return: 1
[DEBUG] Output: <command-line>:0:0: warning: "DEVICE_STDIO_MESSAGES" redefined
[DEBUG] Output: <command-line>:0:0: note: this is the location of the previous definition
[DEBUG] Output: <command-line>:0:0: warning: "DEVICE_SERIAL" redefined
[DEBUG] Output: <command-line>:0:0: note: this is the location of the previous definition
[DEBUG] Output: In file included from /home/adsilva/src/mbed-os/BUILD/mbed/drivers/RawSerial.h:23:0,
[DEBUG] Output:                  from /home/adsilva/src/mbed-os/drivers/RawSerial.cpp:16:
[DEBUG] Output: /home/adsilva/src/mbed-os/BUILD/mbed/drivers/SerialBase.h:166:43: error: 'SerialIrq' has not been declared
[DEBUG] Output:      static void _irq_handler(uint32_t id, SerialIrq irq_type);
[DEBUG] Output:                                            ^~~~~~~~~
[DEBUG] Output: /home/adsilva/src/mbed-os/BUILD/mbed/drivers/SerialBase.h:259:5: error: 'serial_t' does not name a type; did you mean 'serial_s'?
[DEBUG] Output:      serial_t         _serial;
[DEBUG] Output:      ^~~~~~~~
[DEBUG] Output:      serial_s
Traceback (most recent call last):
  File "build.py", line 210, in <module>
    notify=notifier,
  File "/home/adsilva/src/mbed-os/tools/build_api.py", line 1083, in build_mbed_libs
    library_incdirs + incdirs)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 930, in compile_sources
    return self.compile_queue(queue, objects)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 993, in compile_queue
    raise ToolException(err)
ToolException: <command-line>:0:0: warning: "DEVICE_STDIO_MESSAGES" redefined
<command-line>:0:0: note: this is the location of the previous definition
<command-line>:0:0: warning: "DEVICE_SERIAL" redefined
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/adsilva/src/mbed-os/BUILD/mbed/drivers/RawSerial.h:23:0,
                 from /home/adsilva/src/mbed-os/drivers/RawSerial.cpp:16:
/home/adsilva/src/mbed-os/BUILD/mbed/drivers/SerialBase.h:166:43: error: 'SerialIrq' has not been declared
     static void _irq_handler(uint32_t id, SerialIrq irq_type);
                                           ^~~~~~~~~
/home/adsilva/src/mbed-os/BUILD/mbed/drivers/SerialBase.h:259:5: error: 'serial_t' does not name a type; did you mean 'serial_s'?
     serial_t         _serial;
     ^~~~~~~~
     serial_s
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

deece, 2018-11-30:
`mbed-os-5.4.7` builds, but `mbed-os-5.5.0` fails for related (but different) reasons

0xc0170, 2018-11-30:
The problem here is that `DEVICE_` macro must be undefined, not just set to 0. If you look at how serial HAL or SerialBase is protected: `#if defined(DEVICE_SERIAL)` 

This should fix the build problem

deece, 2018-11-30:
Most instances of DEVICE_SERIAL are a plain `#if DEVICE_SERIAL`, so it must be defined. I think the correct fix would be to remove the `defined()` wrappers from the checks in RawSerial.h, Serial.h & SerialBase.h, which was erroneously introduced in commit 26b9a1f6a3

```
$ git grep DEVICE_SERIAL | grep -v DEVICE_SERIAL_ | grep -v Binary
doxyfile_options:                         DEVICE_SERIAL           \
drivers/RawSerial.cpp:#if DEVICE_SERIAL
drivers/RawSerial.h:#if defined (DEVICE_SERIAL) || defined(DOXYGEN_ONLY)
drivers/Serial.cpp:#if DEVICE_SERIAL
drivers/Serial.h:#if defined (DEVICE_SERIAL) || defined(DOXYGEN_ONLY)
drivers/SerialBase.cpp:#if DEVICE_SERIAL
drivers/SerialBase.h:#if defined (DEVICE_SERIAL) || defined(DOXYGEN_ONLY)
drivers/UARTSerial.cpp:#if (DEVICE_SERIAL && DEVICE_INTERRUPTIN)
drivers/UARTSerial.cpp:#endif //(DEVICE_SERIAL && DEVICE_INTERRUPTIN)
drivers/UARTSerial.h:#if (DEVICE_SERIAL && DEVICE_INTERRUPTIN) || defined(DOXYGEN_ONLY)
drivers/UARTSerial.h:#endif //(DEVICE_SERIAL && DEVICE_INTERRUPTIN) || defined(DOXYGEN_ONLY)
features/unsupported/rpc/RpcClasses.h:#if DEVICE_SERIAL
features/unsupported/tests/libs/SerialHalfDuplex/SerialHalfDuplex.cpp:#if DEVICE_SERIAL
features/unsupported/tests/libs/SerialHalfDuplex/SerialHalfDuplex.h:#if DEVICE_SERIAL
hal/serial_api.h:#if DEVICE_SERIAL
platform/mbed_board.c:#if DEVICE_SERIAL
platform/mbed_board.c:#if DEVICE_SERIAL
platform/mbed_retarget.cpp:#if DEVICE_SERIAL
platform/mbed_retarget.cpp:    // Just swallow the data - this is historical non-DEVICE_SERIAL behaviour
platform/mbed_retarget.cpp:#if DEVICE_SERIAL
platform/mbed_retarget.cpp:#else // DEVICE_SERIAL
platform/mbed_retarget.h: *     true and the target has DEVICE_SERIAL;
platform/mbed_retarget.h: *     "platform.stdio-buffered-serial" is false and the target has DEVICE_SERIAL;
platform/mbed_retarget.h: *     target does not have DEVICE_SERIAL.
targets/TARGET_Analog_Devices/TARGET_ADUCM302X/TARGET_ADUCM3029/api/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Analog_Devices/TARGET_ADUCM4X50/TARGET_ADUCM4050/api/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K82F/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_KL27Z/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_KL43Z/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_KL82Z/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_KW24D/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_KW41Z/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K22F/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K24F/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51822_SBKIT/device.h:#define DEVICE_SERIAL           1
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF51/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF51/serial_api.c:#endif // DEVICE_SERIAL
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c:#error UART0 is disabled. DEVICE_SERIAL must also be disabled to continue.
targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c:#endif // DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M2351/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M2351/serial_api.c:#endif  // #if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M451/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M451/serial_api.c:#endif  // #if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M480/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_M480/serial_api.c:#endif  // #if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_NANO100/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_NANO100/serial_api.c:#endif  // #if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_NUC472/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NUVOTON/TARGET_NUC472/serial_api.c:#endif  // #if DEVICE_SERIAL
targets/TARGET_NXP/TARGET_LPC11U6X/objects.h:#if DEVICE_SERIAL
targets/TARGET_NXP/TARGET_LPC11U6X/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NXP/TARGET_LPC82X/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_IMX/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_ONSEMI/TARGET_NCS36510/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_ONSEMI/TARGET_NCS36510/serial_api.c:#endif /* DEVICE_SERIAL  */
targets/TARGET_STM/PeripheralPins.h:#ifdef DEVICE_SERIAL
targets/TARGET_STM/TARGET_STM32F0/serial_device.c:#if DEVICE_SERIAL
targets/TARGET_STM/TARGET_STM32F0/serial_device.c:#endif /* DEVICE_SERIAL */
targets/TARGET_STM/TARGET_STM32F1/serial_device.c:#if DEVICE_SERIAL
targets/TARGET_STM/TARGET_STM32F1/serial_device.c:#endif /* DEVICE_SERIAL */
targets/TARGET_STM/TARGET_STM32F2/serial_device.c:#if DEVICE_SERIAL
targets/TARGET_STM/TARGET_STM32F2/serial_device.c:#endif /* DEVICE_SERIAL */
targets/TARGET_STM/TARGET_STM32F3/serial_device.c:#if DEVICE_SERIAL
targets/TARGET_STM/TARGET_STM32F3/serial_device.c:#endif /* DEVICE_SERIAL */
[... cut ...]
targets/TARGET_Silicon_Labs/TARGET_EFM32/PeripheralNames.h:#if DEVICE_SERIAL
targets/TARGET_Silicon_Labs/TARGET_EFM32/PeripheralPins.h:#if DEVICE_SERIAL
targets/TARGET_Silicon_Labs/TARGET_EFM32/TARGET_EFM32GG11/PeripheralPins.c:#if DEVICE_SERIAL
targets/TARGET_Silicon_Labs/TARGET_EFM32/common/objects.h:#if DEVICE_SERIAL
targets/TARGET_Silicon_Labs/TARGET_EFM32/serial_api.c:#if DEVICE_SERIAL
targets/TARGET_Silicon_Labs/TARGET_EFM32/serial_api.c:#endif //DEVICE_SERIAL
targets/TARGET_WIZNET/TARGET_W7500x/serial_api.c:#if DEVICE_SERIAL
```

0xc0170, 2018-12-03:
There's some inconsistencies here, we will fix it (patch welcome !).

kjbracey-arm, 2018-12-04:
> Most instances of DEVICE_SERIAL are a plain #if DEVICE_SERIAL, so it must be defined.

That's not necessarily so. An undefined macro is substituted with `0` by the preprocessor in a `#if` expression. I'm not actually sure whether those are explicitly defined as '0'.

They probably are, because I've seen enough people adding unnecessary `#if defined` tests in conjunction with `#if` to make me think that at least one compiler issues a warning for undefined macros in `#if` constructs, but I'm not sure.

Will put more comments on the PR.

deece, 2018-12-04:
Wow, I was not expecting that, but it is confirmed in the C 99 standard (http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf), section 6.10.1:
"After all  replacements  due  to  macro  expansion  and  the defined unary operator   have   been   performed,   all   remaining   identifiers   (including   those   lexically identical  to  keywords)  are  replaced  with  the  pp-number 0, and  then  each  preprocessing token  is  converted  into  a  token."

I do think there is some value in keeping things consistent though, Would we be better off making all DEVICE tests ifdefs instead?

deece, 2018-12-05:
@kjbracey-arm Is there a way I can notify the build system of an external fragment to targets.json, or do I need to patch it?

kjbracey-arm, 2018-12-05:
I've never tried it myself, but I've read those docs and seen "The Mbed OS tools now look for custom targets in a file named custom_targets.json in the root of an application and treat custom targets the same as Mbed targets."

So I believe you can put an entry in there that `"inherits"` a target in targets.json and removes the devices.

> I do think there is some value in keeping things consistent though, Would we be better off making all DEVICE tests ifdefs instead?

Possibly, yes.

commit 24dbfb6b (2018-12-20): Remove #ifdefs for DEVICE_SERIAL & friends

DEVICE_SERIAL is always defined (either 0 or 1).
Remove the faulty checks introduces in commit
26b9a1f6a3691e843a49074b64df3aca93d854ae and replace them with
value checks as originally implemented.

Fixes #8913
PR #9163 (2019-01-07): Don't use define checks on DEVICE_FOO macros (partner code)
### Description

Continuation of #8957  (partner components).

The DEVICE_FOO macros are predefined to be 0 or 1. There are a number of instances which incorrectly check if they are defined, rather than checking the value.

This PR addresses most (if not all) of these incorrect checks.

Fixes #8913

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change