ARMmbed/mbed-os #8917: UNO_91H: arm lib heap/stack undefined for ARMCC

0xc0170, 2018-11-30:
### Description

The issue was found in exporters run for 2 different pull request in the last 2 days.

This is the log

```
08:25:15 [UNO_91H:uvision5] .\BUILD\mbed-os-example-blinky.axf: Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from .\build\mbed_boot_arm_std.o).
08:25:15 [UNO_91H:uvision5] .\BUILD\mbed-os-example-blinky.axf: Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from .\build\mbed_boot_arm_std.o).
08:25:15 [UNO_91H:uvision5] .\BUILD\mbed-os-example-blinky.axf: Error: L6218E: Undefined symbol Image$$ARM_LIB_STACK$$ZI$$Base (referred from .\build\mbed_boot_arm_std.o).
08:25:15 [UNO_91H:uvision5] .\BUILD\mbed-os-example-blinky.axf: Error: L6218E: Undefined symbol Image$$ARM_LIB_STACK$$ZI$$Length (referred from .\build\mbed_boot_arm_std.o).
08:25:15 [UNO_91H:uvision5] Not enough information to list load addresses in the image map.
08:25:15 [UNO_91H:uvision5] Finished: 1 information, 0 warning and 4 error messages.
08:25:15 [UNO_91H:uvision5] ".\BUILD\mbed-os-example-blinky.axf" - 4 Error(s), 68 Warning(s).
08:25:15 [UNO_91H:uvision5] Target not created.
08:25:15 [UNO_91H:uvision5] Build Time Elapsed:  00:01:12
08:25:15 [UNO_91H:uvision5] 
08:25:15 [UNO_91H:uvision5] FAILURE building mbed-os-example-blinky UNO_91H uvision5
08:25:15 [UNO_91H:uvision5] ################################################################################
08:25:15 [UNO_91H:uvision5] # Examples compilation summary
08:25:15 [UNO_91H:uvision5] ################################################################################
08:25:15 [UNO_91H:uvision5] #
08:25:15 [UNO_91H:uvision5] ################################################################################
08:25:15 [UNO_91H:uvision5] # Failed build combinations
08:25:15 [UNO_91H:uvision5] ################################################################################
08:25:15 [UNO_91H:uvision5] # mbed-os-example-blinky UNO_91H uvision5
```

This target passed previously  exporters, the very first run (how this target got to master) is here http://mbed-os.s3-eu-west-1.amazonaws.com/builds/exporter/8753/PASS/UNO_91H/uvision5/a2bced7db55d60cea6ac7db33b8975acd63725a2_exporter_build_log_UNO_91H_uvision5.txt

blinky uvision5 all passed. I don't recall any changes since then (was merged 2 weeks ago via https://github.com/ARMmbed/mbed-os/pull/8753).

I tested yesterday master to build with `mbed compile` and also export to uvision5 , all passed on blinky example.

@kyliuxing Please review

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2018-11-30:
@timurh01 @alekla01 Can you review ? 8915 is the one failing, and check the integration PR that passed

From the logs, this one I do not understand (the job number 826 in uvision5 exporters)

```
08:25:10 [UNO_91H:uvision5] Error: hudson.AbortException: script returned exit code 1
08:25:13 [UNO_91H:uvision5] [C:\builds\ws\mbed-os-ci_exporter-uvision5\826] Running shell script
08:25:15 [UNO_91H:uvision5] + tail -n 100 examples/exporter_build_log_UNO_91H_uvision5.log
```

Abort exception first, and then it runs the script and starts compiling?

0xc0170, 2018-11-30:
My test

blinky example, using Mbed OS SHA: `5047ea4f42 - (HEAD -> master, upstream/master) Merge pull request #8889 from kfnta/psa_travis_check` (latest master as of now).

` mbed export -m UNO_91H -i uvision5` - using uvision 5.26 (same update as we use in CI, I updated recently when we updated uvision version in CI)

```
linking...
Program Size: Code=36720 RO-data=2732 RW-data=2584 ZI-data=292324  
".\BUILD\mbed-os-example-blinky.axf" - 0 Error(s), 66 Warning(s).
```

Also some exporters run within the last 2 weeks and this failure can be seen in at least 2 PR (another failure from yesterday http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/builds/exporter/8898/2/uvision5/FAIL/UNO_91H/ )

alekla01, 2018-11-30:
````
08:25:10 [UNO_91H:uvision5] Error: hudson.AbortException: script returned exit code 1
08:25:13 [UNO_91H:uvision5] [C:\builds\ws\mbed-os-ci_exporter-uvision5\826] Running shell script
08:25:15 [UNO_91H:uvision5] + tail -n 100 examples/exporter_build_log_UNO_91H_uvision5.log
08:25:15 [UNO_91H:uvision5] compiling tls_ccm_crypt.c...
08:25:15 [UNO_91H:uvision5] compiling Timer.cpp...
````
"Abort exception first, and then it runs the script and starts compiling?" No
When compilation fails, print the last 100 lines of log (tail -n 100 examples/exporter_build_log_UNO_91H_uvision5.log)

0xc0170, 2019-04-12:
@kyliuxing Is this still correct or can be closed?

kyliuxing, 2019-04-13:
@c0170 It seems to be corrected by commit a814078f0ce58ee0ab6e, I will check it .

0xc0170, 2019-08-26:
@kyliuxing I'll close this as resolved.

closed 2019-08-26 by 0xc0170