ARMmbed/mbed-os #8927: 5.11.0 - RC1 breaks PDMC on some targets.

ashok-rao, 2018-11-30:
### Description

The following targets are affected by this issue:
1. NUMAKER_PFM_M487
2. UBLOX_EVK_ODIN_W2
3. NUMAKER_IOT_M487
4. UBLOX_C030_U201

With 5.10.4 the PDMC tests on these exact boards are passing fine. Logs below:
[test-runner-2018-11-30 09-13-48.txt](https://github.com/ARMmbed/mbed-os/files/2634589/test-runner-2018-11-30.09-13-48.txt)

However, updating ONLY the Mbed OS to 5.11.0-RC1 is breaking the same tests on the same boards. Logs below.
[test-runner-2018-11-30 11-41-36.txt](https://github.com/ARMmbed/mbed-os/files/2634597/test-runner-2018-11-30.11-41-36.txt)

This indicates there is something breaking these targets between the 2 versions. Steps required for reproducing can be shared internally for now.

This also impacts OoB for 5.11.0.
cc: @MarceloSalazar @screamerbg @0xc0170 @cmonr .

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

samchuarm, 2018-12-03:
@ARMmbed/team-nuvoton

0xc0170, 2018-12-03:
I am reviewing the changes between the versions. First thing t hat came to my mind: MPU enablement.

@ashok-rao Can you quickly test one of the failing tests with 5.11rc1, remove `MPU` from `device_has` from `MCU_M480` target  and add `MBED_MPU_CUSTOM` to macros.

0xc0170, 2018-12-04:
@c1728p9 Do you have a device to test this ?

c1728p9, 2018-12-05:
Hi @0xc0170 I don't have either of the Nuvoton boards to test with.

0xc0170, 2018-12-05:
@ashok-rao @ARMmbed/team-nuvoton If anyone can help us narrow down the problem would be great

cyliangtw, 2018-12-05:
@0xc0170 , I can't find any 5.11.x tag in mbed-os master branch, so just pull the last mbed-os. M487-iot log file shows network connect successfully but failed in "`Initialize Simple PDMC`", the root cause seems TLS init HW ECC twice and we are in modifying HW ECC init implementation to fulfill this use case.

[m487-iot-ouput.txt](https://github.com/ARMmbed/mbed-os/files/2648872/m487-iot-ouput.txt)

0xc0170, 2018-12-05:
Use https://github.com/ARMmbed/mbed-os/releases/tag/tools-release-5.11.0-rc1 tag, or can just fetch master as it should be also there (hasn't been yet addressed).

cmonr, 2018-12-05:
@cyliangtw Since we're in the middle of a feature release, the tag won't be added until the release is published.

For now, you can use either the `mbed-os-5.11` branch, or master. Once https://github.com/ARMmbed/mbed-os/pull/8974 comes in, you should only use `mbed-os-5.11` for testing.

Patater, 2018-12-07:
We can probably turn the MPU back on on this target, if the ECC issue is the only issue.

We have root caused the ECC issue and have a fix available at https://github.com/ARMmbed/mbed-os/pull/9005

adbridge, 2018-12-07:
@c1728p9 See Jaeden's comment

cmonr, 2018-12-07:
@adbridge @Patater It'll probably remain off for the rest of OOB since that can be brought in during a patch release. 

@c1728p9 to confirm.

c1728p9, 2018-12-11:
Sounds good to me.

0xc0170, 2018-12-11:
@ashok-rao What is remaining here (for this issue to be closed) ?

ashok-rao, 2018-12-11:
@0xc0170 .. This -> https://github.com/ARMmbed/mbed-os/pull/9020

cmonr, 2018-12-11:
Closing since #9020 is now merged.