ARMmbed/mbed-os #8929: 5.11.0 - RC1 breaks PDMC on some ublox targets.

ashok-rao, 2018-11-30:
### Description

The following targets are affected by this issue:
1. UBLOX_EVK_ODIN_W2
2. UBLOX_C030_U201

With 5.10.4 the PDMC tests on these exact boards are passing fine. Logs below:
[test-runner-2018-11-30 09-13-48.txt](https://github.com/ARMmbed/mbed-os/files/2634861/test-runner-2018-11-30.09-13-48.txt)

However, updating ONLY the Mbed OS to 5.11.0-RC1 is breaking the same tests on the same boards. Logs below.
[test-runner-2018-11-30 11-41-36.txt](https://github.com/ARMmbed/mbed-os/files/2634862/test-runner-2018-11-30.11-41-36.txt)

This indicates there is something breaking these targets between the 2 versions.

Steps to reproduce: (please note that this repo is private to Arm & partners only)

mbed import pelion-enablement
cd pelion-enablement
mbed test -t ARM -n simple-mbed-cloud-client-tests-dev_mgmt* -m <targets_mentioned_above>
Expected:
All tests for connect + update should pass.

Actual:
Connect + update tests fails with either TIMEOUT or ERROR for the targets mentioned above.

This also impacts OoB for 5.11.0.
cc: @MarceloSalazar @screamerbg @0xc0170 @cmonr .

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2018-12-03:
> @cmonr : https://github.com/ARMmbed/pelion-enablement/blob/master/mbed_app.json#L21 .. you can put in your AP's credentials here ..

Ah, my mistake. Was following the instructions verbatum since it was late in the day. Will try it out early tomorrow.

c1728p9, 2018-12-03:
Hi @ashok-rao, @cmonr the error logs attached for the `UBLOX_EVK_ODIN_W2-ARM` and `UBLOX_C030_U201` indicate that a null pointer write was caught by the MPU.

Bit 1 - DACCVIOL - Indicates that there was a data access violation
Bit 7 - MMARVALID - Indicates that MMFAR has valid contents
```
[1543580665.30][CONN][RXD] MMFSR: 00000082
```

The address of the memory location that caused an MPU fault:
```
[1543581892.72][CONN][RXD] MMFAR: 00000000
```

@ashok-rao could you attach the logs from the NUMAKER runs so I could check those as well?

c1728p9, 2018-12-03:
The Odin problems are likely caused by the MPU being enabled. If this is the case then #8920 will fix this.

cmonr, 2018-12-04:
@ashok-rao Please re-test with master, since #8920 has been merged.

Allergies ended up finally hitting me, and I wasn't able to progress on this today. If this is still an issue tomorrow, I can take another look.

adbridge, 2018-12-05:
@ashok-rao @MarceloSalazar Can someone from PE please retest this to see if #8920 has fixed this ?

ashok-rao, 2018-12-06:
RC2 seems to have fixed this on u-blox (with MPU changes).

screamerbg, 2018-12-07:
@c1728p9 Will provide you with test results during the next few hours for these 2 boards.

screamerbg, 2018-12-08:
@c1728p9 this is great! Thanks for your and @juhoeskeli's work on this. Do we have a test case / test to validate this across other partner implementations? It seems that the behaviour wasn't documented or at least wasn't tested against.

c1728p9, 2018-12-11:
Hi @screamerbg, I did a bit of digging to figure out why this wasn't caught in CI. There are two problems lettings this get past CI:
1. The target `UBLOX_C030_U201` is not tested as part of mbed-os CI
2. The mbedtls tests that would have run if part of CI, `tests-mbedtls-multi` and `tests-mbedtls-selftest` do not catch this problem.

This target specific code is not part of the mbed-os HAL or covered by a HAL specification. It is a feature with target specific code residing in `mbed-os/features/mbedtls/targets/TARGET_STM`. @Patater what testing is done for vendor supplied crypto accelerators?

adbridge, 2018-12-13:
@ashok-rao Can this one be closed ? We have #8930 for the underlying issue ?

ashok-rao, 2018-12-13:
Sure. Thanks @adbridge .

closed 2018-12-13 by ashok-rao