ARMmbed/mbed-os #8930: Null pointer dereferenced in libublox-odin-w2-driver.a

c1728p9, 2018-11-30:
### Description

When using WiFi on the UBLOX_EVK_ODIN_W2 with the MPU enabled the device crashes because an invalid pointer is used. This occurs in the function ```cbLPO_startLpoEmulation``` which is located in  ```libublox-odin-w2-driver.a```. Without the MPU enabled writes to a null pointer are ignored on the UBLOX_EVK_ODIN_W2, which is why this has not caused problems in the past.

This failure can be reproduced by running [mbed-os-example-wifi](https://github.com/ARMmbed/mbed-os-example-wifi) on the UBLOX_EVK_ODIN_W2 using a version of mbed-os with the MPU enabled - 7d3e384d7bfaff444dbdef59f500d5eb9785ccb1. Below is a screenshot of this failure occurring:

[image]

Callstack at time of failure:
```
cbLPO_startLpoEmulation() at 0x804a3e0	
initOS() at 0x8047f82	
cbMAIN_initWlan() at 0x80480c8	
OdinWiFiInterface::init() at OdinWiFiInterface.cpp:1,431 0x803e6a2	
OdinWiFiInterface::OdinWiFiInterface() at OdinWiFiInterface.cpp:151 0x803cb84	
WiFiInterface::get_target_default_instance() at default_wifi_interface.cpp:22 0x803f5c2	
WiFiInterface::get_default_instance() at NetworkInterfaceDefaults.cpp:35 0x802fa22	
main() at main.cpp:136 0x8000676	
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

c1728p9, 2018-11-30:
@Team-ublox can you take a look at this?

screamerbg, 2018-12-01:
@c1728p9 It's great that we're adding MPU support, but we haven't communicated to any Mbed Partner that MPU support in drivers (especially binary) is now a requirement. Could we take active steps and ensure that this doesn't happen in the future please?

c1728p9, 2018-12-03:
Hi @screamerbg, don't worry, MPU support in drivers isn't a requirement for partners. The MPU changes were done by the mbed-os HAL team and the mbed-os HAL team is also responsible for fixing any problems caused by this change. In this case the Odin hardfault will be corrected by turning off the MPU. Ublox can then investigate the cause of the null pointer write without being rushed. Regardless, I'll try to keep you in the loop more on changes which are made to a large number of devices, and thus may affect partners.

aqib-ublox, 2019-05-28:
> This failure can be reproduced by running mbed-os-example-wifi on the UBLOX_EVK_ODIN_W2 using a version of mbed-os with the MPU enabled - 7d3e384. Below is a screenshot of this failure occurring:

7d3e384 ref not making sense in this context although i used  https://github.com/ARMmbed/mbed-os-example-wifi and test it for all three tool-chains and we are not getting any hard fault anymore. 

[ARM_logs.txt](https://github.com/ARMmbed/mbed-os/files/3225733/ARM_logs.txt)
[iar_logs.txt](https://github.com/ARMmbed/mbed-os/files/3225734/iar_logs.txt)
[gcc_Arm_logs.txt](https://github.com/ARMmbed/mbed-os/files/3225750/gcc_Arm_logs.txt)

Please verify @c1728p9 , @MarceloSalazar

c1728p9, 2019-05-28:
Hi @aqib-ublox  I mentioned 7d3e384d7bfaff444dbdef59f500d5eb9785ccb1 because it was an mbed-os commit which had the [MPU enabled](https://github.com/ARMmbed/mbed-os/blob/7d3e384d7bfaff444dbdef59f500d5eb9785ccb1/targets/targets.json#L3967) on the UBLOX_EVK_ODIN_W2.

If you add `MPU` to `device_has_add` for the UBLOX_EVK_ODIN_W2 along with removing `MPU_CUSTOM` for this target then the null pointer exceptions should return if the problem had not been fixed already.

aqib-ublox, 2019-05-30:
@c1728p9 

> If you add MPU to device_has_add for the UBLOX_EVK_ODIN_W2 along with removing MPU_CUSTOM for this target then the null pointer exceptions should return if the problem had not been fixed already.

thanks for pointing out yes i am able to reproduce and yes as you pointed out there exists a de-referencing a NULL pointer. It will be resolved in next PR.

0xc0170, 2019-06-06:
> thanks for pointing out yes i am able to reproduce and yes as you pointed out there exists a de-referencing a NULL pointer. It will be resolved in next PR.

When we should expect the fix?

aqib-ublox, 2019-06-11:
@0xc0170 we are planning to have this fix along with ARMC6 https://github.com/ARMmbed/mbed-os/issues/9975. We are currently block on it so cannot commit anything right now.

aqib-ublox, 2019-08-29:
created https://github.com/ARMmbed/mbed-os/pull/11370, please verify @c1728p9

0xc0170, 2019-10-25:
This should be fixed by 11370

closed 2019-10-25 by 0xc0170