ARMmbed/mbed-os #8938: UnicodeDecodeError while synching in greentea test

AmmadRehmat, 2018-12-03:
### Description
While running greentea tests I get following error: 
```

[1543817982.72][COPY][INF] Waiting up to 60 sec for '123602210234651F3D70E614' mount point (current is 'I:')...
        1 file(s) copied.
[1543818032.16][HTST][INF] starting host test process...
[1543818032.60][CONN][INF] starting connection process...
[1543818032.60][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1543818032.60][CONN][INF] initializing serial port listener...
[1543818032.60][PLGN][INF] Waiting up to 60 sec for '123602210234651F3D70E614' serial port (current is 'COM10')...
[1543818032.61][HTST][INF] setting timeout to: 60 sec
[1543818032.62][SERI][INF] serial(port=COM10, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1543818032.62][SERI][INF] reset device using 'default' plugin...
[1543818032.87][SERI][INF] waiting 1.00 sec after reset
[1543818033.87][SERI][INF] wait for it...
[1543818033.90][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1543818033.90][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1543818033.90][CONN][INF] sending preamble '46f32e50-67fe-4dc6-95c1-6ec0ee2d9822'
[1543818033.95][SERI][TXD] {{__sync;46f32e50-67fe-4dc6-95c1-6ec0ee2d9822}}
[1543818033.97][CONN][WRN] UnicodeDecodeError encountered!
[1543818038.96][CONN][INF] Reset the part and send in new preamble...
[1543818038.96][SERI][INF] reset device using 'default' plugin...
[1543818039.21][SERI][INF] waiting 1.00 sec after reset
[1543818040.21][SERI][INF] wait for it...
[1543818040.21][CONN][INF] resending new preamble '86e36da9-0da1-4492-9954-38a95279eb68' after 5.01 sec
[1543818040.25][SERI][TXD] {{__sync;86e36da9-0da1-4492-9954-38a95279eb68}}
[1543818040.26][CONN][WRN] UnicodeDecodeError encountered!
[1543818045.26][CONN][INF] Reset the part and send in new preamble...
[1543818045.26][SERI][INF] reset device using 'default' plugin...
[1543818045.51][SERI][INF] waiting 1.00 sec after reset
[1543818046.51][SERI][INF] wait for it...
[1543818046.51][CONN][INF] resending new preamble '52a2e674-651c-456e-b613-bc91e1da0c3e' after 5.01 sec
[1543818046.55][SERI][TXD] {{__sync;52a2e674-651c-456e-b613-bc91e1da0c3e}}
[1543818046.56][CONN][WRN] UnicodeDecodeError encountered!
[1543818046.57][CONN][WRN] UnicodeDecodeError encountered!
[1543818051.56][CONN][INF] Reset the part and send in new preamble...
[1543818051.56][SERI][INF] reset device using 'default' plugin...
[1543818051.81][SERI][INF] waiting 1.00 sec after reset
[1543818052.81][SERI][INF] wait for it...
[1543818052.81][CONN][INF] resending new preamble '36f3c768-e74b-4d17-8718-5baf8433d740' after 5.01 sec
[1543818052.85][SERI][TXD] {{__sync;36f3c768-e74b-4d17-8718-5baf8433d740}}
[1543818052.86][CONN][WRN] UnicodeDecodeError encountered!
[1543818057.85][CONN][INF] Reset the part and send in new preamble...
[1543818057.85][SERI][INF] reset device using 'default' plugin...
[1543818058.10][SERI][INF] waiting 1.00 sec after reset
[1543818059.10][SERI][INF] wait for it...
[1543818059.10][CONN][INF] resending new preamble '43ae4a92-83be-4d4a-bf52-c662449785b7' after 5.00 sec
[1543818059.14][SERI][TXD] {{__sync;43ae4a92-83be-4d4a-bf52-c662449785b7}}
[1543818059.16][CONN][WRN] UnicodeDecodeError encountered!
[1543818059.17][HTST][ERR] None
[1543818059.17][HTST][WRN] stopped to consume events due to __notify_sync_failed event
[1543818059.17][HTST][INF] __exit_event_queue received
[1543818059.17][HTST][INF] test suite run finished after 26.57 sec...
[1543818059.17][HTST][INF] CONN exited with code: 0
[1543818059.17][HTST][INF] No events in queue
[1543818059.17][HTST][INF] host test result() call skipped, received: sync_failed
[1543818059.17][HTST][WRN] missing __exit event from DUT
[1543818059.17][HTST][INF] calling blocking teardown()
[1543818059.17][HTST][INF] teardown() finished
[1543818059.17][HTST][INF] {{result;sync_failed}}
```

Synch always fails with above error.

Target: UBLOX_EVK_ODIN_W2
ToolChain : GCC_ARM
python version : `2.7.14`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2018-12-03:
I've noticed this error as well in one of CI logs. 

Job mbed-os-ci_greentea-test/459

```
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.46][CONN][RXD] DNS: query "google.com" => "216.58.207.238"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.46][CONN][INF] found KV pair in stream: {{__testcase_start;ASYNCHRONOUS_DNS_NON_ASYNC_AND_ASYNC}}, queued...
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.56][CONN][RXD] DNS: query "youtube.com" => "172.217.22.174"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.56][CONN][RXD] DNS: query "facebook.com" => "31.13.81.36"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.66][CONN][RXD] DNS: query "wikipedia.org" => "91.198.174.192"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.66][CONN][RXD] DNS: query "yahoo.com" => "72.30.35.9"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.77][CONN][RXD] DNS: query "instagram.com" => "50.16.58.231"
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.86][CONN][WRN] UnicodeDecodeError encountered!
03:55:24 [NUCLEO_F746ZG armcc] mbedgt: :50::FAIL: Expression Evaluated To FALSE
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.87][CONN][RXD] :50::FAIL: Expression Evaluated To FALSE
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.97][GLRM]0217000034544e45001000018aa900292011000097969900 - socketio resource_appeared
03:55:24 [NUCLEO_F746ZG armcc] [1543722923.99][CONN][WRN] UnicodeDecodeError encountered!
```

@ARMmbed/mbed-os-test @ARMmbed/mbed-os-tools

bridadan, 2018-12-03:
@0xc0170 That error can happen, it indicates that there were non-unicode bytes read from the serial port. I would expect this would typically come from a device sending corrupted data (or extra garbage bytes in a buffer?).

aashishc1988, 2018-12-03:
@AmmadRehmat Do you mind sharing the test script that is causing this issue? It would be interesting to see what is going on that is causing the sync itself to fail.

AmmadRehmat, 2018-12-04:
@aashishc1988 I am getting this error for every test scrip that i used with mbed test not some particular script.

@bridadan  Please refer to my reply to ashish and look into it if possible. I have mentioned my python version and I am using windows 10.

bridadan, 2018-12-04:
@AmmadRehmat can you tell us which version of Mbed OS you're using? If you're working off the master branch, which commit hash are you working from? Thanks.

aashishc1988, 2018-12-04:
@AmmadRehmat I tried running tests on the same platform with the exact version of tools you provided and I couldnt reproduce it. As @bridadan said, can you tell the version of MbedOs you're using, and if possible, also tell me the name of the specific test you are running which has this problem. I know you said every test script has the problem, so unless you have ran all of them, that info can be helpful. Reason I am asking for test script is, in the past there were certain tests that were able to demonstrate this issue on certain platforms. 
Right now we are just looking for ways to reproduce this issue to really understand what is going on, so any help would be appreciated :)

AmmadRehmat, 2018-12-05:
@aashishc1988  Reinstalling python solved the issue, for reference I was running mbed-os-tests-mbedmicro-rtos-mbed-basic , on mbed-os-5.9

bridadan, 2018-12-05:
Interesting, @AmmadRehmat did you keep the same version of Python? 2.7.14?

naveenkaje, 2018-12-05:
closing as @AmmadRehmat reported that reinstalling python solved the issue.

closed 2018-12-05 by naveenkaje

aashishc1988, 2018-12-06:
@AmmadRehmat It would be useful to know what version of python you installed that solved this issue. May be there is something that we are missing or have not understood on that front.

AmmadRehmat, 2018-12-07:
@aashishc1988 Its the same version of python previously installed 2.7.14

bridadan, 2018-12-07:
Ok, if it reappears do let us know! Thanks!