ARMmbed/mbed-os #8943: NXP LPC1768 is padding ethernet frames to 64 bytes

KariHaapalehto, 2018-12-03:
### Description
Frame lenght should be 64 bytes, including headers and checksum bytes. 
LPC176X driver is padding ethet frames to 64 bytes before checksum bytes. So adding checksum bytes  will increased total size to 68.

This error has been tested with UBLOX_C027 using Wireshark to capture log.
Attached WireShark log from greentea network emac tests shows that first 3 packets are length of 100 bytes and they are transmitted correctly. Packets 4-6, the original lenght was 60 bytes, so the data block should be 32 bytes, not 36 bytes.
Because of this padding error, the network emac test aren't passing with UBLOX_C027.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

KariHaapalehto, 2018-12-03:
[ublox_c027.zip](https://github.com/ARMmbed/mbed-os/files/2638940/ublox_c027.zip)
WireShark log file attached

kjbracey-arm, 2018-12-03:
@mikaleppanen - how's your memory? Didn't we have some case where excess padding was unavoidable? I seem to recall a test being relaxed rather than fixing it.

TuomoHautamaki, 2018-12-04:
@ARMmbed/team-nxp please have a look

adbridge, 2021-06-07:
As there has been no update on this for over 2 years I am closing it. Please re-open if this is still a valid issue.

closed 2021-06-07 by adbridge