ARMmbed/mbed-os #8948: psa_security.c breaks LoRaWAN builds

hasnainvirk, 2018-12-03:
### Description

Bugs and un-flagged code in psa_crypto.c breaks all the builds for LoRaWAN.
The problem is that the LoRaWAN builds remove MBEDTLS_GCM_C by undefining it in its Mbedtls configuration file. psa_crypto.c seems to use MBEDTLS_CCM_C or MBEDTLS_GCM_C and at some places it is correctly flagged. However, there are many places where it is not properly flagged, e.g., 
https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/mbed-crypto/platform/TARGET_PSA/COMPONENT_PSA_SRV_IMPL/psa_crypto.c#L3149 here the check is needed for GCM not for CCM.

https://github.com/ARMmbed/mbed-os/blob/master/features/mbedtls/mbed-crypto/platform/TARGET_PSA/COMPONENT_PSA_SRV_IMPL/psa_crypto.c#L3265 here the GCM is not flagged at all.   

This is a blocker for LoRaWAn builds. Please fix asap.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

hasnainvirk, 2018-12-03:
It seems there are plenty of issues related to this issue and there is a fix PR open https://github.com/ARMmbed/mbedtls/pull/2241/commits/7d6f48a058b0c93e95383fa5d9a0b8da62b617b6 @0xc0170 What is the status on this PR ? Our builds are blocked because of this.

0xc0170, 2018-12-03:
I believe this should fix it https://github.com/ARMmbed/mbed-os/pull/8926 (can you test?)

hasnainvirk, 2018-12-03:
@0xc0170 I am not sure why we didn't see it before. Let me try with the PR you mentioned.

hasnainvirk, 2018-12-03:
@0xc0170 I can confirm that the #8926 fixes the issue.

adbridge, 2018-12-03:
Looks like a duplicate of #8893

adbridge, 2018-12-04:
@hasnainvirk Could you please check if the TLS update #8926 fixes this ?

adbridge, 2018-12-05:
@hasnainvirk @AnttiKauppila  any update ?

AnttiKauppila, 2018-12-05:
@adbridge All our Lora CI jobs are passing so this seems to have fixed the issue

0xc0170, 2018-12-05:
I'll close this as resolved

closed 2018-12-05 by 0xc0170