ARMmbed/mbed-os #8954: nRF52832 power consumption during sleep

drahnr, 2018-12-03:
### Description

    - What target does this relate to? NRF52DK
    - What toolchain (name + version) are you using? gcc 7.x
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? mbed-cli 1.7.5
    - What is the SHA of Mbed OS (git log -n1 --oneline)? 5.10 c53d51fe9220
    - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?) Not yet.

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

During supposedly low power states, the power consumption is still higher than specified.

Allocating a string is enough to trigger this. There are other triggers too.

@janjongboom I will detail this out as soon as I have detailed measurements.

pan-, 2018-12-03:
Hi @drahnr, 

Could you post a sample code that reproduces the issue and indicate the profile used ? 

Side note: GCC7.x is **not** supported.

janjongboom, 2018-12-04:
@marcuschangarm Have you seen this?

desmond-blue, 2019-07-31:
@drahnr, since we haven't heard from you for a while, we would like to check if you still need help on this issue? Thanks.

closed 2019-07-31 by drahnr