ARMmbed/mbed-os #8962: [NRF51832, NRF52832, SDT51822B, SDT52832B] PWM issues

heejungPark, 2018-12-04:
### Description
Compiler : GCC_ARM
OS : latest mbed-os

We are testing to control PWM on SDT51822B or SDT52832B.
It is same with NRF51822, NRF52832 each other.
But it is not working when we added this main.cpp on latest mbed-os.
It make sure two NRF board is not working too.
Please check main.cpp and serial messages as below.

```
#include "mbed.h"

/* Serial */
#define BAUDRATE 9600
Serial serial_pc(USBTX, USBRX, BAUDRATE);

/* Servo motor */
PwmOut pwmOut0(PWM0);
// PwmOut pwmOut0(PWM1);
// PwmOut pwmOut0(PWM2);
// PwmOut pwmOut0(PWM3);

int main(void) {
    serial_pc.printf("< Sigma Delta Technologies Inc. >\n\r");

    pwmOut0.period_ms(20);

    serial_pc.printf("start\n");
    pwmOut0.write(0.100);
    wait(2);

    while(true) {
        pwmOut0.write(0.100);
        wait(2);
        pwmOut0.write(0.075);
        wait(2);
        pwmOut0.write(0.050);
        wait(2);
        pwmOut0.write(0.075);
        wait(2);
    }
}
```

Serial message of SDT51822B:
```
< Sigma Delta Technologies Inc. >

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x1B843
Error Value: 0x0
Current Thread: Id: 0x200048E4 Entry: 0x1D281 StackSize: 0x1000 StackMem: 0x20004928 SP: 0x200057F0 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
nrf failure at .\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_11\drivers_nrf\timer\nrf_drv_timer.c:149
```

Serial message of SDT52832B:
```

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x37359
Error Value: 0x0
Current Thread: Id: 0x2000546C Entry: 0x338B9 StackSize: 0x1000 StackMem: 0x200054B0 SP: 0x200063E0 
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
nrf failure at .\mbed-os\targets\TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_14_2\drivers_nrf\pwm\nrf_drv_pwm.c:192
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

andrewc-arm, 2018-12-05:
Hi, HeeJung

It's just for our Arm internal ticketing purpose. We will update you when there is a progress about it.

Thanks,

TacoGrandeTX, 2018-12-19:
@heejungPark You appear to be observing issue https://github.com/ARMmbed/mbed-os/issues/7743.  It has been fixed in Mbed OS v5.9.7 and newly released v5.11 (see  https://github.com/ARMmbed/mbed-os/pull/7779/commits/e1d50193b04c5f5856fe1c607b822dbd89a1461f).  It's not clear why this didn't make it into v5.10.4, but I tested this on a nRF52_DK (using LED1 as there is no PWM0) for v5.11 and no run-time error is generated.  Please confirm this for your targets too.