ARMmbed/mbed-os #8963: Enable split heap in SRAM1 and SRAM2 on Nucleo L476RG

maxgerhardt, 2018-12-04:
### Description

The Nucleo L476RG (STM32L476RG) has two SRAM banks (see [datasheet](https://www.st.com/resource/en/datasheet/stm32l476rg.pdf) chapter 3.5): 
* SRAM 1, 96 kB at 0x20000000
* SRAM 2, 32 kB at 0x10000000

Using the mbed-os's heap statistics (`MBED_HEAP_STATS_ENABLED`) and the minimal application at the end, I see that the RAM used by the firmware is 10740 bytes out of 131072. The available for size for heap is 87560 bytes. If I add the available heap size and the 10740 bytes usage I get 96*1024 bytes, i.e. the SRAM1 size. Thus the heap uses only the leftover size in SRAM1. 

I however noticed that the implementation explicitly looks for a "two memory region model" and implements a special `_sbrk` wrapper function for it. 

See https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/platform/mbed_retarget.cpp#L1198-L1208 and https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c#L35-L63. 

I have verified that this code is being compiled in and executed. However this code doesn't expand the available heap size since it has been activate from the start and the heap still only uses SRAM2.

My question is: Does mbed-os have code for expanding the heap across SRAM1 and SRAM2? How can I activate that? Do I have to implement it myself?

As a workaround to get a bigger available heap size in a firmware I moved objects in an SRAM2 region to free up SRAM1. (`__attribute__((__section__(".stack_dummy")))`, see linker script  https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/device/TOOLCHAIN_GCC_ARM/STM32L476XX.ld#L150-L156). Is this OK or does this destroy / overwrite / overlap with some stack memory?

Application:

```cpp
#include "mbed.h"
#include "rtos.h"

Serial pc(USBTX, USBRX, "debug", 921600); //TX, RX

void debug_heap() {
    mbed_stats_heap_t heap_info;
    mbed_stats_heap_get( &heap_info );
    printf("\n\tBytes allocated currently: %d", (int) heap_info.current_size);
    printf("\n\tMax bytes allocated at a given time: %d", (int) heap_info.max_size);
    printf("\n\tCumulative sum of bytes ever allocated: %d", (int) heap_info.total_size);
    printf("\n\tCurrent number of bytes allocated for the heap: %d", (int) heap_info.reserved_size);
    printf("\n\tCurrent number of allocations: %d", (int) heap_info.alloc_cnt);
    printf("\n\tNumber of failed allocations: %d\n", (int) heap_info.alloc_fail_cnt);
    printf("\n");
}

/* number of bytes allocated in the exhaustion test */
#define HEAP_ALLOC_SIZE 256

void exhaust_heap() {
    /* allocate as much memory as we can until we get an error */
    uint32_t allocedBytes = 0;
    while(true) {
        uint8_t* data = (uint8_t*) malloc(HEAP_ALLOC_SIZE);

        if(!data)
            break;
        allocedBytes += HEAP_ALLOC_SIZE;
    }
    printf("Heap exhaustion after allocating %d bytes\n", (int) allocedBytes);
}

int main() {
#ifdef TWO_RAM_REGIONS
    pc.printf("TWO RAM REGIONS defined\n");
#else
    pc.printf("TWO RAM REGIONS undefined\n");
#endif

#if defined(__GNUC__)
    pc.printf("__GNUC__ defined\n");
#endif

#if !defined(__CC_ARM)
    pc.printf("No __CC_ARM defined\n");
#endif

#if !defined(__ARMCC_VERSION)
    pc.printf("__ARMCC_VERSION not defined\n");
#endif

    debug_heap();
    exhaust_heap();
    debug_heap();

	return 0;
}
```

Output:

```
TWO RAM REGIONS defined
__GNUC__ defined
No __CC_ARM defined
__ARMCC_VERSION not defined

	Bytes allocated currently: 428
	Max bytes allocated at a given time: 428
	Cumulative sum of bytes ever allocated: 428
	Current number of bytes allocated for the heap: 87560
	Current number of allocations: 1
	Number of failed allocations: 0

Heap exhaustion after allocating 80896 bytes

	Bytes allocated currently: 82348
	Max bytes allocated at a given time: 82348
	Cumulative sum of bytes ever allocated: 82348
	Current number of bytes allocated for the heap: 87560
	Current number of allocations: 318
	Number of failed allocations: 1
```

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

maxgerhardt, 2018-12-11:
@0xc0170 sure you linked the right team? It gives a 404 for me and there hasn't been a review for a week..

0xc0170, 2018-12-11:
Yes, not certain why it gives 404 but it is (checked just now).

adustm, 2018-12-11:
Hi,
No 404 on my side neither, thanks for the reminder.

> My question is: Does mbed-os have code for expanding the heap across SRAM1 and SRAM2? How can I activate that? Do I have to implement it myself?

As you've seen SRAM1 and SRAM2 are not contiguous on this particular device 
Mbed code was created for contiguous heap and stack sections.
ST team has modified it so that is can support noncontiguous heap and stack sections and free space for the heap memory on SRAM1.

We’ve never thought about splitting heap over several noncontiguous memory sections, you are the first to ask.
I have the conviction that it would complicate the heap memory alloc/free a lot (need to make an allocation trial on memory 1 , then on memory 2 ... need to keep pointers for both memories ...).

If you feel it easy to write, not risky to use, not runtime consuming, available for the 3 toolchains... etc etc...  please share your code and we will be happy to test it on ARM / GCC_ARM and IAR toolchains for you.

> As a workaround to get a bigger available heap size in a firmware I moved objects in an SRAM2 region to free up SRAM1. (__attribute__((__section__(".stack_dummy"))), see linker script [...]

Well, I don't see any problem with that. It should work, thanks to https://github.com/ARMmbed/mbed-os/blob/2fd0c5cfbd83fce62da6308f9d64c0ab64e1f0d6/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/device/TOOLCHAIN_GCC_ARM/STM32L476XX.ld#L165
```
    __StackLimit = __StackTop - SIZEOF(.stack_dummy);
    PROVIDE(__stack = __StackTop);
    /* Check if stack exceeds RAM2 limit */
    ASSERT((ORIGIN(SRAM2)+LENGTH(SRAM2)) >= __StackLimit, "SRAM2 overflow")
```
@LMESTM @0xc0170 do you agree with this workaround ? Are you aware of any limitation of using stack_dummy memory ? 

Kind regards

maxgerhardt, 2018-12-11:
OK, thanks for the answer. If I have free time I might invest some time in implementing this for mbed-os. I mean, FreeRTOS has had split heap support with `heap5.c` for [4 years already](https://github.com/jameswalmsley/FreeRTOS/blob/master/FreeRTOS/Source/portable/MemMang/heap_5.c), so it should be possible.

0xc0170, 2018-12-11:
> BTW regarding the 404. You won't get a 404 since you are part of this team. Every outsider to this project will get a 404.

Yes, the team members are not visible publicly. 

Regarding referenced heap5, it's own allocator , isn't it? We use the default that toolchain provide. We will need to check how this can be handled (span heap over non adjecent memory regions) for various toolchains. 

cc @ARMmbed/mbed-os-core

deepikabhavnani, 2018-12-11:
Yes heap5 is FreeRTOS own allocator, and it gives mechanism to select different types of allocators. 
Currently we do not support external allocators, and heap span over adjacent memory regions is not supported either.

Moving objects to other RAM partitions is the right approach. ISR Stack size in Mbed OS 5 is  1K and is at the end. In case of GCC_ARM heap memory should be 4K aligned, hence you will have additional 4K of heap by moving stack. Since SRAM2 is 32KB, consider moving other sections to to SRAM2 as well (bss / data). 

Some targets use single RAM for heap and other for everything else (vectors / data / bss / stack). Not sure if you can have vectors in SRAM, but moving `bss` to `SRAM2` should be feasible. Main thread stack of 4K and other thread stacks are in `BSS`

closed 2018-12-11 by maxgerhardt