ARMmbed/mbed-os #8967: TEST_IGNORE() macro is not handled properly

alzix, 2018-12-04:
### Description
In some cases we need to ignore/skip current test as we the preconditions are not meet for testing the feature.

There is a dedicated macros in `unity.h` for that special purpose:
- `TEST_IGNORE_MESSAGE(message)`
- `TEST_IGNORE()`
When this macro is called current test execution aborted and we skip to a next test.

Summary tables printed by `mbedgt` does not expose information about test being ignored.
For example - i modified TESTS/mbed_hal/critical_section/main.cpp. Added TEST_IGNORE_MESSAGE() macro in the test body.

The resulting summary table looks like this:
```text
mbedgt: test suite report:
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
| target       | platform_name | test suite                      | result | elapsed_time (sec) | copy_method |
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
| K64F-GCC_ARM | K64F          | tests-mbed_hal-critical_section | OK     | 14.23              | default     |
+--------------+---------------+---------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+--------------+---------------+---------------------------------+-----------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                      | test case                         | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+---------------------------------+-----------------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-mbed_hal-critical_section | Test critical section nested lock | 1      | 0      | OK     | 0.1                |
| K64F-GCC_ARM | K64F          | tests-mbed_hal-critical_section | Test critical section single lock | 1      | 0      | OK     | 0.08               |
+--------------+---------------+---------------------------------+-----------------------------------+--------+--------+--------+--------------------+
```
As we do not have a column for ignored test cases i would expect that result will be marked as `IGNORED` instead of `OK`.

Hiding ignored tests is dangerous as we can skip/hide some tests and not not notice that.

### Issue request type

    [ ] Question
    [X] Enhancement
    [X] Bug

cmonr, 2018-12-04:
Thanks for the investigation @alzix!

@davidsaada @ARMmbed/mbed-os-test-team @ARMmbed/mbed-os-maintainers Fyi.

davidsaada, 2018-12-04:
@alzix Thanks for reporting this issue. This was the original intention, but it was a bit too complicated with the current way utest was implemented. But I agree this is the desired way.

jeromecoutant, 2019-01-04:
Hi
What is the status of this issue ?
It is really not easy to read test reports with only OK status,
and then we need to check manually if each test is really OK or only SKIPPED...
Thx

davidsaada, 2019-01-06:
Don't think it's currently assigned to anyone. Believe it should fall under @ARMmbed/mbed-os-test-team's responsibility.

0xc0170, 2019-01-07:
> Don't think it's currently assigned to anyone. Believe it should fall under @ARMmbed/mbed-os-test-team's responsibility.

This was assigned internally to test team.

closed 2020-07-16 by ciarmcom