ARMmbed/mbed-os #8977: Exporting to MCUXpresso IDE results in error

fluidblue, 2018-12-05:
### Description

When trying to export a project to MCUXpresso using mbed-cli, e.g. with

    mbed export -i mcuxpresso -m LPC11U68

the command results in an error.
The error states that LPC11U68 is not supported by mcuxpresso, however `mbed export --supported` shows that this microcontroller is supported on mcuxpresso.

Some details: mbed-cli v1.8.3 was used on the project mbed-os-example-blinky.

This also happens for other LPC microcontrollers, e.g. LPC11U35_401.

Why is the export failing?
How to solve it?

Appendix:
Error message from mbed-cli:

	[mbed] Working path "/Users/max/Desktop/test/mbed-os-example-blinky" (program)
	Scan: mbed-os-example-blinky
	usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
	                  [-L] [-S [{matrix,ides}]] [--update-packs] [-E]
	                  [--build BUILD_DIR] [--source SOURCE_DIR] [-D MACROS]
	                  [--profile PROFILE] [--app-config APP_CONFIG] [-z]
	                  [--ignore IGNORE]
	project.py: error: LPC11U68 not supported by mcuxpresso
	[mbed] ERROR: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python" returned error.
	       Code: 2
	       Path: "/Users/max/Desktop/test/mbed-os-example-blinky"
	       Command: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python -u /Users/max/Desktop/test/mbed-os-example-blinky/mbed-os/tools/project.py -i mcuxpresso -m LPC11U68 --source ."
	       Tip: You could retry the last command with "-v" flag for verbose output
	---

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

naveenkaje, 2018-12-05:
Hi @fluidblue 
Thanks for reporting this issue. We will look at this issue and get back to you on this thread.

fluidblue, 2018-12-17:
I did some debugging:

	Traceback (most recent call last):
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli/mbed-os-example-blinky/mbed-os/tools/project.py", line 374, in main
	    ignore=options.ignore
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli/mbed-os-example-blinky/mbed-os/tools/project.py", line 155, in export
	    ignore=ignore
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 274, in export_project
	    toolchain.config.load_resources(resources)
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli/mbed-os-example-blinky/mbed-os/tools/config/__init__.py", line 1107, in load_resources
	    raise NotSupportedException("Target does not support mbed OS 5")
	NotSupportedException: Target does not support mbed OS 5

With this information found out that the LPC11U68 is only compatible with mbed OS 2.

**I think the above error information ("Target does not support mbed OS 5") should be added to the mbed-cli output ("LPC11U68 not supported by mcuxpresso").**

With the additional error information I also realized that there are two versions of the blinky example: [mbed OS 5 blinky](https://github.com/ARMmbed/mbed-os-example-blinky) and [mbed OS 2 blinky](https://mbed.org/teams/mbed/code/mbed_blinky/). For boards only supporting mbed OS 2 (e.g. LPC11U68), the mbed OS 2 blinky example must be used.

naveenkaje, 2019-03-08:
Fix merged.

closed 2019-03-08 by naveenkaje