ARMmbed/mbed-os #8984: Failed to build UBLOX_EVK_ODIN_W2 using IAR 8.2

andresag01, 2018-12-05:
### Description

(This issue mirrors [this](https://github.com/ARMmbed/mbed-os-example-tls/issues/137) issue in mbed-os-example-tls)

The application [mbed-os-example-tls/authcrypt](https://github.com/ARMmbed/mbed-os-example-tls/tree/master/authcrypt) fails to build when using IAR 8.2. The command used to build the software is:
```
mbed deploy
mbed compile -m UBLOX_EVK_ODIN_W2 -t IAR
```

The error returned is:

```
Compile [100.0%]: error.c
Link: authcrypt
Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
          built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li009]: runtime model conflict: Module
          cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
          must be 'C++', but module authcrypt.o has the value 'C++14'. Modules
          compiled with C++14 are not link-compatible with modules that are
          compiled with older C++.
Error[Li060]: module "hci_vendor_cmd.o(libublox-odin-w2-driver.a)" contains C++
          code built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
          C++ code built for use with the (E)C++98 library. This code is not
          link compatible with the new (C++14) library.
Warning[Lt009]: Inconsistent wchar_t size
            cb_main.o(libublox-odin-w2-driver.a) and 120 other objects have
                      wchar_t size 16 bits
            authcrypt.o and 1084 other objects, some of them in dlpp7M_tl_fc.a
                      and 207 other libraries have wchar_t size 32 bits
[ERROR] Error[Li060]: module "cb_main.o(libublox-odin-w2-driver.a)" contains C++ code
          built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li009]: runtime model conflict: Module
          cb_main.o(libublox-odin-w2-driver.a) specifies that '__CPP_Language'
          must be 'C++', but module authcrypt.o has the value 'C++14'. Modules
          compiled with C++14 are not link-compatible with modules that are
          compiled with older C++.
Error[Li060]: module "hci_vendor_cmd.o(libublox-odin-w2-driver.a)" contains C++
          code built for use with the (E)C++98 library. This code is not link
          compatible with the new (C++14) library.
Error[Li060]: module "cb_ticker_wrapper.o(libublox-odin-w2-driver.a)" contains
          C++ code built for use with the (E)C++98 library. This code is not
          link compatible with the new (C++14) library.
Warning[Lt009]: Inconsistent wchar_t size
            cb_main.o(libublox-odin-w2-driver.a) and 120 other objects have
                      wchar_t size 16 bits
            authcrypt.o and 1084 other objects, some of them in dlpp7M_tl_fc.a
                      and 207 other libraries have wchar_t size 32 bits

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt"
       Command: "c:\python27\python.exe -u C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\make.py -t IAR -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\IAR -c"
       Tip: You could retry the last command with "-v" flag for verbose output
```
Interestingly, when I try to get more debug information by running the command:
```
mbed compile -t IAR -m UBLOX_EVK_ODIN_W2 -c --verbose
```
I get a different error which looks like a different problem in mbed-cli:
```
Compile [ 54.5%]: thread_mdns.c
Compile: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.0\arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 -Oh --enable_restrict --cpu Cortex-M4F --thumb --dlib_config DLib_Config_Full.h --vla --diag_suppress=Pe546 -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x200000 -DTARGET_STM32F439 -DDEVICE_SPI=1 -DDEVICE_EMAC=1 -D__MBED__=1 -DTARGET_STM32F439xx -D__FPU_PRESENT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DMBEDTLS_SHA1_C -DDEVICE_RTC=1 -DTARGET_CORDIO_ODIN_W2 -DTARGET_STM32F4 -D__CMSIS_RTOS -DCB_CHIP_WL18XX -DHSE_STARTUP_TIMEOUT=5000 -DWLAN_ENABLED -DMBEDTLS_CONFIG_HW_SUPPORT -DTARGET_MODULE_UBLOX_ODIN_W2 -DDEVICE_USTICKER=1 -DHSE_VALUE=24000000 -DSUPPORT_80211D_ALWAYS -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DDEVICE_WIFI=1 -DMBED_BUILD_TIMESTAMP=1544052737.61 -DFEATURE_BLE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_LPTICKER=1 -DTOOLCHAIN_IAR -DTARGET_STM32F439xI -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCB_FEATURE_802DOT11W -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DTARGET_CORDIO -DMBEDTLS_ARC4_C -DDEVICE_SERIAL_FC=1 -DMBEDTLS_MD4_C -DCB_FEATURE_802DOT11R -DMBEDTLS_DES_C -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -DTARGET_STM32F439ZI -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_UBLOX_EVK_ODIN_W2 -DDEVICE_CAN=1 -DCB_INTERFACE_SDIO -DMBEDTLS_MD5_C -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DARM_MATH_CM4 -DTARGET_STM_EMAC -f .\BUILD\UBLOX_EVK_ODIN_W2\IAR\.includes_9c0efdeaa9bdb601583e2772cbd9b80b.txt --preinclude=.\BUILD\UBLOX_EVK_ODIN_W2\IAR\mbed_config.h --dependencies BUILD\UBLOX_EVK_ODIN_W2\IAR\mbed-os\features\nanostack\sal-stack-nanostack\source\6LoWPAN\Thread\thread_mdns.d -l BUILD\UBLOX_EVK_ODIN_W2\IAR\mbed-os\features\nanostack\sal-stack-nanostack\source\6LoWPAN\Thread\thread_mdns.s.txt -o BUILD\UBLOX_EVK_ODIN_W2\IAR\mbed-os\features\nanostack\sal-stack-nanostack\source\6LoWPAN\Thread\thread_mdns.o .\mbed-os\features\nanostack\sal-stack-nanostack\source\6LoWPAN\Thread\thread_mdns.c
[DEBUG] Return: 0
Traceback (most recent call last):
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\make.py", line 71, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\build_api.py", line 538, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\toolchains\__init__.py", line 413, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\toolchains\__init__.py", line 467, in compile_queue
    res['command']
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\toolchains\__init__.py", line 578, in compile_output
    self.notify.debug("Output: %s"% error_line)
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\notifier\__init__.py", line 72, in debug
    self.notify({'type': 'debug', 'message': message})
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\notifier\term.py", line 69, in notify
    msg = self.print_notify_verbose(event)
  File "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\notifier\term.py", line 114, in print_notify_verbose
    return "[DEBUG] {message}".format(**event)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u20ac' in position 345: ordinal not in range(128)
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt"
       Command: "c:\python27\python.exe -u C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os\tools\make.py -t IAR -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\IAR -c -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

Other useful information:

- What toolchain (name + version) are you using?
```
C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt>iccarm --version
IAR ANSI C/C++ Compiler V8.20.1.14183/W32 for ARM
```
- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
```
C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt>C:\Python27\Scripts\mbed --version
1.8.3
```
- What is the SHA of mbed-os-example-tls (git log -n1 --oneline)?
```
C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt>git log -n1 --oneline
b16a5136 (HEAD -> master, tag: mbed-os-5.10.4, origin/master, origin/HEAD) Merge pull request #215 from cmonr/master
```
- What is the SHA of Mbed OS (git log -n1 --oneline)?
```
C:\Users\Administrator\Downloads\mbed-os-example-tls\authcrypt\mbed-os>git log -n1 --oneline
2fd0c5cfb (HEAD -> mbed-os-5.10, tag: mbed-os-5.10.4, tag: latest, origin/mbed-os-5.10) Merge pull request #8775 from ARMmbed/release-candidate
```
I am using Windows 10.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cmonr, 2018-12-06:
@andresag01 Thank you for the incredibly detailed log.

Mbed OS is currently not compatible with IAR 8.*:
- https://os.mbed.com/docs/v5.10/tools/configuring-mbed-cli.html
- https://os.mbed.com/docs/v5.10/tools/exporting.html

andresag01, 2018-12-06:
@cmonr Thanks for the clarification! I should have looked at the documentation more carefully. However, I still have a couple of questions:
* If we already know that Mbed OS is not compatible with certain versions of compilers, why doesnt the tool give a warning message when I try to build with an unsupported toolchain? For example, if I try running mbed-cli with Python3 it gives a clear message saying "this version does not work" and this really helps usability.
* Why does mbed-cli fail with a `UnicodeError` when I enable verbose output? (please refer to the backtrace in the issue description)

cmonr, 2018-12-06:
Interesting. We should be checking the version: https://github.com/ARMmbed/mbed-os/pull/7247/files
It's possible that the IAR check may need to be updated, since the version string in IAR 8.* is slightly different.

> Why does mbed-cli fail with a UnicodeError when I enable verbose output? (please refer to the backtrace in the issue description)

That will require a bit more investigation. We try to squash unicode errors wherever we find them, but this one looks new. I can see if I can reproduce it with a different compiler, but it's possible it could be due to only IAR 8.*.

closed 2019-02-05 by ciarmcom

cmonr, 2019-02-11:
Closed since we don't support IAR8 yet. Feel free to reopen once support is introduced and if this still doesn't work.

Fyi @SenRamakri Jira/GitHub syncing...