ARMmbed/mbed-os #8991: Netsocket tests suite faults target DISCO_L475VG_IOT01A

ConradBraam, 2018-12-06:
### Description

    Running network netsocket test suite crashes the L475 Discovery target (wifi).
toolchain GCC_ARM and mbed cli project
/master
network_test ( revision in the current branch)
`- mbed-os (#830132548196)

1. Follow steps in https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md 
2. Do not fully set up a wifi router yet.
3. compile and run only the wifi set:
mbed test -n mbed-os-tests-network-wifi -v
4. Observed a repeatable crash in the first test and in all the other tests in the set - attached hard fault logs showing crash in wifi_connect_nocredential( x ) function

The same application builds, and does not crash on Odin-W2 target. Raising this as a possible issue in the tests themselves, because I'm only checking the tests, not the examples, which must validate separately at some point.

[logs.txt](https://github.com/ARMmbed/mbed-os/files/2653738/logs.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

ConradBraam, 2018-12-06:
It looks pretty much like a default-stack /config issue on the surface not being handled well in the tests perhaps.

ConradBraam, 2018-12-06:
See also : https://github.com/ARMmbed/mbed-os/issues/8055 for more clues.

0xc0170, 2018-12-12:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-12-12:
Hi

Your test is crashing at test case 2 'WIFI-CONNECT-NOCREDENTIALS', but it is already failed at step 1 'WIFI-CONSTRUCTOR' ...

Could you precise your mbed_app json file ?

ConradBraam, 2018-12-12:
Hmmm, looks like user error, my JSON file was like so:

"target_overrides": {
		"DISCO_L475VG_IOT01A":{
			"target.network-default-interface-type": "WIFI",	
            "nsapi.default-wifi-ssid": "\"Zaphodikus\"",
            "nsapi.default-wifi-password": "\"<password>\"",
            "nsapi.default-wifi-security": "WPA2"
		},
               "*": {
			"target.network-default-interface-type": "ETHERNET"
		}
	}
That does not parse like you might expect it to, because the * gets to overwrite the discovery and sets the setting to default ETHERNET. I should have copy-pasted the JSON instead of writing from memory and trying to add multiple targets.
When I moved the "*" wildcard up above the target, and then re-checked yesterday afternoon, and verified different toolchains as well, that resolves the error. I have raised a separate issue around the JSON-ness of the file.  So I guess we can close this one off.