ARMmbed/mbed-os #8992: Support for Cortex-M33F-NS

mmahadevan108, 2018-12-06:
### Description
Cortex-M33F-NS is not implemented correctly in the mbed-os tools folder for the gcc and arm toolchains. The overall implementation assumes Cortex-M33, Cortex-M33-NS and needs to be updated for Cortex-M33F, Cortex-M33F-NS

### Issue request type
    [ ] Question
    [ ] Enhancement
    [X] Bug

deepikabhavnani, 2018-12-06:
@mmahadevan108  - Please provide more detail of what seems incorrect
Cortex-M33 / Cortex-M33-NS - maps to cortex-M33 core without any optional features. https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L399

In case core selected is Cortex-M33F-NS - floating point is enabled.
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L420

mmahadevan108, 2018-12-06:
1. In the loop from 391-406
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L391

It is looking for 'f' at the end however, cortex-m33f-ns does not have it at the end. 

2. The below does not seem to be considering Cortex-M33F
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L423

There are other instances in the gcc file.

mmahadevan108, 2018-12-06:
Core value is set below:
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L399

deepikabhavnani, 2018-12-06:
> The below does not seem to be considering Cortex-M33F
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L423

Agree to this secure side build is not considering the floating point, but for this we will need input from PSA / TF-M team. @dreemkiller - Should we have floating point enabled on secure side?

deepikabhavnani, 2018-12-06:
> Core value is set below:
https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/arm.py#L399

yes.. And do you see anything incorrect with that?

mmahadevan108, 2018-12-06:
I am seeing the below build failures when core is set to cortex-m33+nodsp, works when core is set to cortex-m33.
___________________________
Error: L6366E: BUILD/LPC55S69/ARMC6/mbed-os/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC/us_ticker.o attributes
are not compatible with the provided attributes .
Object BUILD/LPC55S69/ARMC6/mbed-os/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_LPC/us_ticker.o contains Build Attr
ibutes that are incompatible with the provided attributes.
    Tag_FP_arch = Use of the ARM v8-A FP ISA was permitted, but only citing registers D0-D15, S0-S31 (=8)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
Not enough information to list the image map.
Finished: 2889 information, 1 warning and 722 error messages.

[mbed] ERROR: "c:\python27\python.exe" returned error.
____________________________

deepikabhavnani, 2018-12-06:
I have few questions.
1. Error is during secure build / non-secure build
2. What is the core set for secure and non-secure builds

mmahadevan108, 2018-12-06:
"LPC55S69": {
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M33F-NS",
        "default_toolchain": "ARMC6",
        "supported_toolchains": ["ARMC6", "GCC_ARM"],
        "extra_labels": [
            "NXP",
            "MCUXpresso_MCUS",
            "LPCXpresso",
            "LPC"
        ],
        "is_disk_virtual": true,
        "macros": [
            "CPU_LPC55S69JBD100_cm33_core0",
            "ARM_MATH_CM33",
            "MBED_TZ_DEFAULT_ACCESS=1",
            "MBED_FAULT_HANDLER_DISABLED",
            "FSL_RTOS_MBED"
        ],
        "inherits": ["Target"],
        "detect_code": ["0236"],
        "device_has": [
            "USTICKER",
            "RTC",
            "I2C",
            "I2CSLAVE",
            "INTERRUPTIN",
            "PORTIN",
            "PORTINOUT",
            "PORTOUT",
            "SERIAL",
            "SLEEP",
            "SPI",
            "SPISLAVE",
            "STDIO_MESSAGES",
            "FLASH"
        ],
        "release_versions": ["5"],
        "device_name": "LPC55S69_core0",
        "bootloader_supported": true
    },

mmahadevan108, 2018-12-06:
Error was seen during non-secure build

deepikabhavnani, 2018-12-06:
Also do you have optional DSP is enabled in your target?

mmahadevan108, 2018-12-06:
Yes, I believe the below lines should do it.
#if ((__FPU_PRESENT == 1) && (__FPU_USED == 1))
  SCB->CPACR |= ((3UL << 10*2) | (3UL << 11*2));    /* set CP10, CP11 Full Access */
#endif /* ((__FPU_PRESENT == 1) && (__FPU_USED == 1)) */

  SCB->CPACR |= ((3UL << 0*2) | (3UL << 1*2));    /* set CP0, CP1 Full Access (enable PowerQuad) */

  SCB->NSACR |= ((3UL << 0) | (3UL << 10));   /* enable CP0, CP1, CP10, CP11 Non-secure Access */

deepikabhavnani, 2018-12-06:
Please see if https://github.com/deepikabhavnani/mbed-os/commit/ebe5e5fb8fab638da8bf509b1b73349e05a5672c commit works for you, select core as `Cortex-M33FD-NS`

mmahadevan108, 2018-12-06:
My guess is the DSP is being used by the power library.

deepikabhavnani, 2018-12-12:
PR #8995 resolves this

closed 2018-12-12 by deepikabhavnani