ARMmbed/mbed-os #8999: NVStore storage area reservation

pauluap, 2018-12-07:
### Description

I am evaluating the NVStore feature on mbed-os 5.8.3.  I'm trying to figure out how the flash area is reserved.  I do see the configuration options for `NVSTORE_AREA_1_ADDRESS` and friends, but I don't see anything related to compile-time placement or any configuration to the linker scripts.  Do the current examples implicitly rely on code not being present in the specified locations?

In the device I'm using, the STM32F469, not all of the sectors are the same size, so I'd like to optimize the selection of a specific area of FLASH.  I was expecting to need to construct a section in the linker script to reserve the area used by NVStore.  Would you have any examples of that or do you know of a different way to enforce that the linker does not place code in the reserved area?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

cmonr, 2018-12-07:
@ARMmbed/mbed-os-storage

pauluap, 2018-12-07:
Thank you for the confirmation.

closed 2018-12-07 by pauluap