ARMmbed/mbed-os #9000: [OoB_5.11.0-RC2]: Debugging with exporting vscode_gcc_arm failed

desmond-blue, 2018-12-07:
Steps to reproduce:
1. mbed export -i vscode_gcc_arm -m K64F --profile debug
2. On Debug tab, click Play icon.
3. An error message occurred 

```
mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/fsl_common.o -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group, ...) failed.
make (e=87): The parameter is incorrect.

make.exe[1]: *** [mbed-os-example-blinky.elf] Error 87
make.exe: *** [all] Error 2
```

    [ ] Question
    [ ] Enhancement
    [ ] Bug

adbridge, 2018-12-07:
@desmond-blue Thanks for the issue, for future reference please do not remove the Descriptions and Issue request type headers from the template and also please put a x in the correct issue type. I have corrected this for you on this occasion.

theotherjimmy, 2018-12-13:
@desmond-blue This is a known issue, I think. You're going to have the same problem with the `make` exporters.

0xc0170, 2019-08-14:
Should be fixed via #9596

closed 2019-08-14 by 0xc0170