ARMmbed/mbed-os #9002: LPC17XX USBSerial hangs on receive

kallaspriit, 2018-12-07:
### Description

USBSerial on LPC1768 hangs when receiving first message.

- Sometimes works after resetting the board. Once it works it keeps working for next restarts until the board is power-cycled.
- Does not work when using PlatformIO with VSCode on Windows 10 with the latest dependencies.
- Works properly when using online mbed compiler (but this uses older version of USBSerial).
- *Also works when using PlatformIO with VSCode on Windows 10 when using older version of features/unsupported/USBDevice/targets/TARGET_NXP\USBHAL_LPC17.cpp copied from mbed online compiler*.

Thus my conclusion is that the modifications made to USBHAL_LPC17.cpp have broken it.

Test procedure:

- Wire up USBSerial as normal (needs 15k resistors to ground for D- and D+).
- Compile the code against the latest USBDevice as it is currently in the github repository.
- Upload code to mbed LPC1768.
- Attach the USB device (tried both Windows 10 computer and Android phone).
- Power cycle the board.
- Connect to the LPC1768 from the USB device (it should start receiving the message every 500ms).
- Attempt to send a message to the board (should hang - LED1 stops blinking and does not respond any more).
- Resetting the board multiple times and attempting to send more messages sometimes works.

Here's my test code:

```c++
#include "mbed.h"
#include "USBSerial.h"

// usb device configuration (same as defaults)
const uint16_t USB_VENDOR_ID = 0x1f00;
const uint16_t USB_PRODUCT_ID = 0x2012;
const uint16_t USB_PRODUCT_RELEASE = 0x0001;

// log serial configuration
const PinName LOG_SERIAL_TX_PIN = USBTX;
const PinName LOG_SERIAL_RX_PIN = USBRX;
const int LOG_SERIAL_BAUDRATE = 115200;

// setup log and usb serial
Serial logSerial(LOG_SERIAL_TX_PIN, USBRX, LOG_SERIAL_BAUDRATE);
USBSerial usbSerial(USB_VENDOR_ID, USB_PRODUCT_ID, USB_PRODUCT_RELEASE, false);

// setup status LEDs
DigitalOut loopStatusLed(LED1);
DigitalOut connectionStatusLed(LED2);

// timer for blinking a LED to show that the main loop is still active
Timer blinkTimer;

// handles usb serial receive callback
void handleSerialRx()
{
  // proxy all available characters to log serial
  while (usbSerial.available() > 0)
  {
    logSerial.printf("%c", usbSerial.getc());
  }
}

int main()
{
  // listen for usb serial data
  usbSerial.attach(&handleSerialRx);

  blinkTimer.start();

  while (true)
  {
    // run this code every 500ms
    if (blinkTimer.read_ms() >= 500)
    {
      // toggle the loop status led, making it blink when the main loop is running properly
      loopStatusLed = !loopStatusLed;

      // check for usb serial connection status
      if (usbSerial.connected())
      {
        // indicate connected state
        connectionStatusLed = 1;

        // send messages to both log and usb serials
        logSerial.printf("connected\n");
        usbSerial.printf("hey\n");
      }
      else
      {
        // indicate disconnected state
        connectionStatusLed = 0;

        logSerial.printf("disconnected\n");
      }

      blinkTimer.reset();
    }
  }
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

c1728p9, 2019-02-26:
Hi @kallaspriit the USB code in `features/unsupported/USBDevice` is no longer supported. I recommend you use https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-usb-device instead.

0xc0170, 2019-02-27:
@kallaspriit You can also ask on developer portal, someone might have similar experience.

> Hi @kallaspriit the USB code in `features/unsupported/USBDevice` is no longer supported

This will be closed now.

closed 2019-02-27 by 0xc0170