ARMmbed/mbed-os #9006: Network Socket Test Fail: TCPSOCKET_THREAD_PER_SOCKET_SAFETY FAIL

tz-arm, 2018-12-07:
### Description
- Type: Bug
- Related issue: none
- Priority: Major
 
---------------------------------------------------------------
## Bug
 
**Target**
K64F + Quectel BC28 Arduino Shield
 
**Toolchain:**
GCC_ARM
 
**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]
 
**mbed-cli version:** 
1.8.3
 
**mbed-os sha:**
$ mbed ls
mbed-os-example-cellular (#06111ecda36d)
`- mbed-os (#103c9cb509c7, tag: mbed-os-5.11.0-rc1)
 
**DAPLink version:**
Version: 0246
 
**Expected behavior**
 
All test cases passed in netsocket tests suite.
 
**Actual behavior**

 TCPSOCKET_THREAD_PER_SOCKET_SAFETY FAIL.
 
```
[1544180637.49][CONN][RXD] >>> 'TCPSOCKET_ECHOTEST_NONBLOCK': 1 passed, 0 failed
[1544180637.49][CONN][RXD]
[1544180637.49][CONN][INF] found KV pair in stream: {{__testcase_finish;TCPSOCKET_ECHOTEST_NONBLOCK;1;0}}, queued...
[1544180637.51][CONN][RXD] >>> Running case #3: 'TCPSOCKET_OPEN_CLOSE_REPEAT'...
[1544180637.51][CONN][RXD] >>> 'TCPSOCKET_OPEN_CLOSE_REPEAT': 1 passed, 0 failed
[1544180637.51][CONN][RXD]
[1544180637.51][CONN][INF] found KV pair in stream: {{__testcase_start;TCPSOCKET_OPEN_CLOSE_REPEAT}}, queued...
[1544180637.51][CONN][INF] found KV pair in stream: {{__testcase_finish;TCPSOCKET_OPEN_CLOSE_REPEAT;1;0}}, queued...
[1544180637.52][CONN][RXD] >>> Running case #4: 'TCPSOCKET_OPEN_LIMIT'...
[1544180637.52][CONN][RXD] [round#00] unable to open new socket, error: -3005
[1544180637.52][CONN][RXD] [round#00] 7 sockets opened
[1544180637.52][CONN][INF] found KV pair in stream: {{__testcase_start;TCPSOCKET_OPEN_LIMIT}}, queued...
[1544180637.54][CONN][RXD] [round#01] unable to open new socket, error: -3005
[1544180637.54][CONN][RXD] [round#01] 7 sockets opened
[1544180637.54][CONN][RXD] >>> 'TCPSOCKET_OPEN_LIMIT': 1 passed, 0 failed
[1544180637.54][CONN][RXD]
[1544180637.54][CONN][INF] found KV pair in stream: {{__testcase_finish;TCPSOCKET_OPEN_LIMIT;1;0}}, queued...
[1544180637.55][CONN][RXD] >>> Running case #5: 'TCPSOCKET_THREAD_PER_SOCKET_SAFETY'...
[1544180637.55][CONN][RXD] AT+NSOCR=STREAM,6,53714,1
[1544180637.55][CONN][INF] found KV pair in stream: {{__testcase_start;TCPSOCKET_THREAD_PER_SOCKET_SAFETY}}, queued...
[1544180637.58][CONN][RXD]
[1544180637.58][CONN][RXD] 5
[1544180637.58][CONN][RXD]
[1544180637.59][CONN][RXD] OK
[1544180637.59][CONN][RXD] AT+NSOCO=5,52.215.34.155,7
[1544180637.62][CONN][RXD]
[1544180637.62][CONN][RXD] OK
[1544180637.63][CONN][RXD] AT+NSOCR=STREAM,6,53761,1
[1544180637.65][CONN][RXD]
[1544180637.68][CONN][RXD] +CME ERROR: 4
[1544180637.68][CONN][RXD] network error -3012
[1544180637.68][CONN][RXD] :73::FAIL
[1544180948.63][HTST][INF] test suite run finished after 480.81 sec...
[... 55 lines cut ...]
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_SEND_REPEAT              | 0      | 0      | SKIPPED | 0.0                |
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_SEND_TIMEOUT             | 0      | 0      | SKIPPED | 0.0                |
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_THREAD_PER_SOCKET_SAFETY | 0      | 0      | ERROR   | 0.0                |
+--------------+---------------+-----------------------------+------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 9 SKIPPED / 4 OK / 1 ERROR
mbedgt: completed in 507.68 sec
mbedgt: exited with code 1
[mbed-14724] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\Workspace\mbed_examples\mbed-os-example-cellular"
       Command: "mbedgt --test-spec .\BUILD\tests\K64F\GCC_ARM\test_spec.json -n mbed-os-tests-netsocket-tcp -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```
 
**Steps to reproduce**
 
Follow the instruction in netsocket test document:
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

tz-arm, 2018-12-07:
The Quectel BC28 module only support 7 sockets, but the default max socket number in NSAPI is 10: 
```
socketstats.h:
#ifndef MBED_CONF_NSAPI_SOCKET_STATS_MAX_COUNT
#define MBED_CONF_NSAPI_SOCKET_STATS_MAX_COUNT      10
#endif
```
Intend to define it in mbed_app.json and try again.

SeppoTakalo, 2018-12-12:
`nsapi.socket-stats-max-count` is only value that defines how many sockets the statistics API keeps in memory.

It is not a limit of how many simultaneous sockets the device supports.

LwIP by default supports 4, but can be configured to larger amount.
ESP8266 supports only 4.

This testcase requires only 2.

adbridge, 2018-12-13:
@SeppoTakalo Could you please confirm whether this is a real issue or not and if it needs to go into known issues ?

tz-arm, 2019-02-19:
Close the issue and will discuss with cellular team how to improve the socket configuration in cellular stack.

closed 2019-02-19 by tz-arm