ARMmbed/mbed-os #9011: LoRaWAN: _can_cancel_tx not reset

einglis, 2018-12-07:
### Description
I'll confess, I don't have a thorough explanation of how I arrived at my discovery, suffice to say the result is pretty clear cut.

In general terms, I'm testing what happens when I fail to get message confirmations, cancel messages while they are still retrying, and occasionally disconnecting and reconnecting.

What I was observing is that after a rejoin, I was unable to cancel messages, even under circumstances where I'd expect to be able to.  I tracked this down to `_can_cancel_tx` in `class LoRaMac` being false when I'd have expected it to be true.

I found it wasn't being reset with ~everything else in `LoRaMac::reset_mac_parameters()`.  The following did the trick for me:

    void LoRaMac::reset_mac_parameters(void)
    {
        _is_nwk_joined = false;
        _can_cancel_tx = true;  // <----
        ...

I realise it's incumbent on me to provide a better explanation and test case, but I thought I'd get the ball rolling in the meantime.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

einglis, 2018-12-07:
Update:
My current working hypothesis is that after one or more failed joins (then a successful join) messages cannot be cancelled until at least one has succeeded.
Which is still pretty woolly, so I'll continue...

einglis, 2018-12-10:
A new week and a fresh pair of eyes...

I've done a more thorough, side-by-side comparison of my test cases, and I think I'm just a victim of timing.    A join failure shifts the duty-cycle hold-offs just long enough to break my next send attempt, and it's nothing to do with the stack, per-se.

I'll leave the bug open for now in case anyone thinks the `_can_cancel_tx` should be reset /anyway/, but I no longer believe it's causing my problem.

hasnainvirk, 2018-12-31:
@einglis I agree with you. I am confident that resetting the _can_cancel_tx flag is not an issue as if there is a scheduled transmission in the pipe that stack will not let you cancel the transmission. And once it is done with transmission, it will flip the flag itself. So there is no need to reset it in the reset_mac_params() API. 
I believe you can close the issue.

closed 2018-12-31 by ciarmcom