ARMmbed/mbed-os #9016: SAMD21J18A Target Support

loverdeg-ep, 2018-12-07:
### Description
I have a product running bare metal on SAMD21J18A.
Software maintenance is a nightmare and costing customer money.

Would like to propose to customer that we port over to MbedOS.

### Issue request type

    [ ] Question
    [x ] Enhancement
    [ ] Bug

cmonr, 2018-12-07:
@janjongboom Fyi.

@loverdeg-ep I'm not sure I follow. Is this a request for ARM to port the SAMD21 MCU to Mbed OS?

janjongboom, 2018-12-08:
@loverdeg-ep I'm working on the SAML21J18B target (https://github.com/ARMmbed/mbed-os/pull/8658), I assume it'd be relatively similar to getting the SAMD21 working. But no plans on my side to also include the SAMD21.

loverdeg-ep, 2018-12-10:
@cmonr Sorry, my shorthand made that confusing.

I meant I would like to propose to the customer that I port their application from baremetal to Mbed OS.

Mbed OS support for SAMD21J18A would need to happen for me to make this proposal.

Request is not for anybody in particular to do so. Though I was hoping @janjongboom might already have plans for it.

I might do it myself if I can find the opportunity to review @janjongboom's other SAM ports.

janjongboom, 2018-12-11:
@loverdeg-eg, great! I've already added some patches to add EDBG to mbed-ls so greentea tests can run, happy to share other porting details with you.

loverdeg-ep, 2018-12-11:
If I start this effort I will obviously need to run tests locally to start.
In order to get the board into automated testing on your end, will the [SAM D21 Xplained Pro](https://www.microchip.com/Developmenttools/ProductDetails/ATSAMD21-XPRO#additional-summary) board (ATSAMD21-XPRO) be a sufficient board target?

janjongboom, 2018-12-19:
@loverdeg-ep I think so. I'm using SAML Xplained Pro, and that can run tests fine.

closed 2019-10-31 by ciarmcom