ARMmbed/mbed-os #9017: [OoB_5.11.0-RC2]: Debugging with exporting vscode_gcc_arm failed

1155105943, 2018-12-07:
### Description   
Steps to reproduce:   
1. mbed export -i vscode_gcc_arm -m K64F --profile debug
2. Open folder from Visual Studio Code 
3. Debug -> Start debugging
4. Error messages occurred   
```
The preLaunchTask 'make.exe' terminated with exit code 1.
```

5. If I modify the  "preLaunchTask": "make.exe" as  "preLaunchTask": "" , the following error messages occur
 ```
'C:\Users\eriche03\OOB\mbed-os-example-blinky_back\...\mbed-os-example-blinky.elf' does not exist
```

6. If I first compile the project by CLI and change the path of mbed-os-example-blinky.elf as correct path, the following error messages occur
```
Unable to start debugging. Debug server process failed to initialize.
```

7. And the path of MIDebuggerPath is wrong on the file launch.json, I modified it before compile.

Environment:
Windows 10
Visual Studio Code
  
### Pull request type
```
[ ] Question
[ ] Enhancement
[x] Bug
```

cmonr, 2018-12-07:
@janjongboom

TacoGrandeTX, 2019-03-13:
@1155105943 Is this still causing you problems?  If so - can you check that the make utility is in your path?  If it is in the path, then you might be hitting an issue with the line length being too long as discussed here:  https://github.com/ARMmbed/mbed-os/issues/6335. 
 
As for the debugging issue, do you see any output from the "Debug Console" tab?

0xc0170, 2019-08-14:
@1155105943 I'll close this due to inactivity. If updated, can be reopened.

closed 2019-08-14 by 0xc0170