ARMmbed/mbed-os #9027: Network Socket Test Fail: TCPSOCKET_THREAD_PER_SOCKET_SAFETY Fail

tz-arm, 2018-12-10:
### Description
- Type: Bug
- Related issue: none
- Priority: Major
 
---------------------------------------------------------------
## Bug
 
**Target**
K64F + Quectel BC28 Arduino Shield
 
**Toolchain:**
GCC_ARM
 
**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]
 
**mbed-cli version:** 
1.8.3
 
**mbed-os sha:**
$ mbed ls
mbed-os-example-cellular (#06111ecda36d)
`- mbed-os (#103c9cb509c7, tag: mbed-os-5.11.0-rc1)
 
**DAPLink version:**
Version: 0246

**Expected behavior**
 
All test cases passed in netsocket tests suite.
 
**Actual behavior**
 
TCPSOCKET_THREAD_PER_SOCKET_SAFETY FAIL by system hang up.  See log file below:
 
```
***** IRQ blinky uVisor example *****
[1544427945.50][CONN][RXD] 3,52.215.34.155,7,601,4E443439354738363A3E4C4752333F4D303A4D53483953525748433C364457524C3249574843333B5335384938593E374550533D4E3A55494B303B3D504757403B515A353B3E55413E4F314448363E3D374E32564249323E5A544D4F45504F5346373253444333353849484731353B4F504F454F4D57353A4F46334E32394B303A45453043575536353F564A44393C304C30435842363448595540575830594239513C4E3C3B53343050533559454C3E43583335345350394F4F5A32493B4256514332554F4F3D4444534D473F5754564A4D3C3E42413B4D4F52304B333E313F485A583D5646544E57305339495044563B57584D51303A4A363734553F5545504D334F4137434F30474E3247344D57464C3038333633574B3D3238574357305A3E4F3B3F493D485438383D524B325A354B4D3C5732463E3E3C53394E384B543A5A5045554A56335042443A464A40554D35505044503E44443D3A4453394744344056463E3132593B4F433330583C485059343041384F41503C54393350525854583B39535132313B33373C5232373E524F36593B36375A474B58453F5432443D32474D57413252414B3432434F5543354E59533D4132373857584D3D47414B5250554E4C4D3F54433D344352344E564E444F383A33594B35463052304C3F5341574D4E5A39574230394C4C3133484254323635583D4B3D3531444330344E4D4441495A314C3B583B4856493A5A52485048303A3559423938564C54433F3353333F3038504D46324E3448324C323C3435445A55503B323456414B43504A41435452323E3D5438503A363348564B53303F53433657433E345052,0
[1544427945.50][CONN][RXD]
[1544427945.50][CONN][RXD] OK
[1544427946.30][CONN][RXD] AT+NSOSD=3,501,..[0]
[1544427946.57][CONN][RXD]
[1544427946.57][CONN][RXD] 3,501
[1544427946.59][CONN][RXD]
[1544427946.59][CONN][RXD] OK
[1544427946.59][CONN][RXD] AT+NSORF=3,501
[1544427946.60][CONN][RXD]
[1544427946.62][CONN][RXD] OK
[1544427949.08][CONN][RXD]
[1544427949.09][CONN][RXD] +NSONMI:3,501
[1544427949.09][CONN][RXD] AT+NSORF=3,501
[1544427949.11][CONN][RXD]
[1544427950.19][CONN][RXD] 3,52.215.34.155,7,501,5A3A54594B363F43555351503B3B37593B344E33384553373744503D3C3B4C3637413A583A4A5352564B4A434E5A5A433C4B35315940544F4A345149443F3D39423C514A5743563F51363F4248545330474443345236373E553C4741553D30424546485A51383E324E39554F4650453431415052373F313E4D3850573D554C5A444D4D3F58573D4854323358374B4445344F4A3E55474F5A3B4748554D353934303F4E4F5032414E3D4C45565156304E304D3C56363A463830403140594940513640365847484B50304E45533531505A42484A333B58344D594D393146353A4A463333535342443653545954333230344D384540504047454248524B363C4533594E4253383944434A3E32543C3B3952393B3E3B514951574C483C3F47503F3953454D54484547353337355A4C525353413B57513B4932575256334E4738493C44545343383037353E474C5A4A554C303A4B3F503436395751454641575A57494D574E463A34464532583B3A5A3D494A4C3353403B4A305151434A3A4842563040555337423B494B4045535555404B564D434256423940474E4F534B4A4C395536333C5233353B5732454A4054454549535A4E544B364B3D4E35535245483538373854454B385748464D3F4B49434D3843335558513A353658364F553E344D51523D42434B3758493437554A32425441504A373358,0
[1544427950.19][CONN][RXD]
[1544427950.19][CONN][RXD] OK
[1544427950.77][CONN][RXD] AT+NSOSD=3,401,..[0]
[1544427951.05][CONN][RXD]
[1544427951.06][CONN][RXD] 3,401
[1544427951.06][CONN][RXD]
[1544427951.07][CONN][RXD] OK
[1544427951.07][CONN][RXD] AT+NSORF=3,401
[1544427951.08][CONN][RXD]
[1544427951.09][CONN][RXD] OK
[1544427953.14][CONN][RXD]
[1544427953.16][CONN][RXD] +NSONMI:3,401
[1544427953.16][CONN][RXD] AT+NSORF=3,401
[1544427953.17][CONN][RXD]
[1544427954.04][CONN][RXD] 3,52.215.34.155,7,401,495A463451304F345A41434645535640503044564C46305553493A394C52484F45543C39553E5332584F48353C393A46584B4539474F414E4E47414F3946443E4B3959424D493B3A59414E385A314952304041583035403B40555A4A5632444D4D4B3E32523E454B5236374B30535252503B3C3359584A563F4449563B3A303C444751575555355131403F3E3B4438303A38385A314344433955423E4B4856353D4058463C54504736325935305833344B36433C453E434E54595A463252353544513733565A4A30484237553E5451464B483443483443524D31474C515851373B595341314E4C454E4B325539483E51384A3E44303A42343D3851374C593E47564936385254585855313B51403547383E3F4B4B31495A3E3A4A54403343515A4135404142384E513836313D3434414E4F484C573747484D4848544B3333494C4D5A3F454852533841574C383C37364B4A3D444B3835563A3B41523142573F38473D52465043323E404D53403E3E34503243393D425053583A3D344B40584033363A4D51553F5258503D34374643543D35,0
[1544427954.05][CONN][RXD]
[... 114 lines cut ...]
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_SEND_REPEAT              | 0      | 0      | SKIPPED | 0.0                |
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_SEND_TIMEOUT             | 0      | 0      | SKIPPED | 0.0                |
| K64F-GCC_ARM | K64F          | mbed-os-tests-netsocket-tcp | TCPSOCKET_THREAD_PER_SOCKET_SAFETY | 0      | 0      | ERROR   | 0.0                |
+--------------+---------------+-----------------------------+------------------------------------+--------+--------+---------+--------------------+
mbedgt: test case results: 9 SKIPPED / 1 OK / 1 ERROR
mbedgt: completed in 507.76 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\Workspace\mbed_examples\mbed-os-example-cellular"
       Command: "mbedgt --test-spec .\BUILD\tests\K64F\GCC_ARM\test_spec.json -n mbed-os-tests-netsocket-tcp -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

**Steps to reproduce**

Follow the instruction in netsocket test document:
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

mhiljane, 2018-12-10:
Terence, please consult Ari Parkkila and Mirela Chirica. Did the cellular example pass but now one of the socket test are failing?

tz-arm, 2018-12-11:
@mhiljane sorry that I did not have chance to run network socket greetea test before.
@AriParkkila @mirelachirica, could you share if any comments about if thread safe in cellular stack, maybe particular for BC95? thanks!

closed 2019-05-23 by ciarmcom