ARMmbed/mbed-os #9031: ARM Cortex-M4 (Nuvoton NUC472) not booting with External RAM on EBI Interface

Ritesh236, 2018-12-10:
ARM Cortex-M4 (Nuvoton NUC472) not booting with External RAM on EBI Interface

Hi,

We are working on ARM Cortex-M4 (Nuvoton NUC472) based SoC as per our requirement.We are facing one critical issue like board is not booting when we enabled and used extrenal RAM on EBI.

We have ported following BSP package with below toolchain for NUC472.

    BSP version : 3.03.000 (Downloaded from https://github.com/OpenNuvoton/NUC472_442BSP)
    Tool-chain Version : arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release).

We have created bare framework with above BSP and we are able to start the board with existing ld script provided in BSP with internal RAM support.

Now, We have added the support for external RAM of 512KB on EBI in linker script as per our requirement to move few libraries into external RAM.So, We have added one test library and placed in external ram to validate and test external RAM with our bare build.
But, Board does not starts if we make any variable as extern to use from library which is placed into external RAM.

It is working fine If we are putting that library into internal RAM then board gets up and running without any stuck up issue.

Please find following CFLAGS and LDFLAGS options which we are using into our compilation support.

CFLAGS :
        -mcpu=cortex-m4
        -mthumb
        -mfpu=fpv4-sp-d16
        -mfloat-abi=hard
        -g2
        -Wall
        -c
        -DNUC442RI8AE
        -D__FPU_USED -std=c99
        -ffunction-sections -fdata-sections
        -fstrict-volatile-bitfields
        -fno-short-enums

LDFLAGS :
        -Wl,--gc-sections
        -Wl,-static
        -Xlinker --no-enum-size-warning
        -Xlinker --wrap=_malloc_r
        -Xlinker --wrap=_free_r
        -Xlinker --wrap=_calloc_r
        -Xlinker --wrap=_realloc_r
        -Wl,--start-group
        -lgcc -lc -lm -lnosys
        -Wl,--end-group

Please find following list of libraries which we have moved into external RAM

    test library
    FreeRTOS Heap Memory

Please find attached sample multi RAM linker script (for reference) and our own sample linker script which we have used into our example

Also, Please find below code execution flow of our sample project on which external RAM on EBI is not working
    
    1) On Board Boot Up, Linker script is calling Reset_Handler from which _start is called
    2) after that main function will be called from main.c file of BSP
    3) Then it will create Hardware Setup, EBI Init, Data Copy and init heap section
    4) After that we have created one sample task to test freeRTOS functionality

So, check above details with linker script and let me know if need any more details from my side regarding same.

Sample multi RAM linker script (for reference)
[multi-ram.txt](https://github.com/ARMmbed/mbed-os/files/2663181/multi-ram.txt)

Our own sample linker script
[gcc_arm_ext_ram.txt](https://github.com/ARMmbed/mbed-os/files/2663184/gcc_arm_ext_ram.txt)

cmonr, 2018-12-11:
> FreeRTOS Heap Memory

Closing as this repository is for the Mbed OS project, but it's obvious that Mbed OS is not being used. Either contact OpenNuvoton or the Nuvoton support team.

closed 2018-12-11 by cmonr

0xc0170, 2018-12-11:
This was already opened on OpenNuvoton repositories, should be redirected there