ARMmbed/mbed-os #9038: Feature: ADC_TEMP & ADC_VREF definitions for U-blox ODIN W2

screamerbg, 2018-12-10:
### Description

It would be great if ADC_TEMP and ADC_VREF can be defined for U-blox ODIN W2, similar to other ST based targets - https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_NUCLEO_F439ZI/PinNames.h#L216

I'd like to use this as part of my device health IoT project.

CC @ARMmbed/team-ublox @andreaslarssonublox 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

screamerbg, 2019-03-28:
@aqib-ublox Any progress on this?

aqib-ublox, 2019-03-29:
@screamerbg not yet will target in next next or week following next

0xc0170, 2019-08-14:
> @screamerbg not yet will target in next next or week following next

@ARMmbed/team-ublox Was this addressed or not yet?

aqib-ublox, 2019-08-20:
@0xc0170 https://github.com/ARMmbed/mbed-os/pull/11266 PR created please verify if it works

aqib-ublox, 2019-08-29:
@0xc0170  shouldn't it be close?

closed 2019-08-29 by 0xc0170