ARMmbed/mbed-os #9039: Feature: ADC_TEMP and ADC_VREF for U-blox C030 U201 and other C030 based boards

screamerbg, 2018-12-10:
### Description

It would be great if ADC_TEMP and ADC_VREF can be defined for U-blox C030 U201 (and all other C030 boards), similar to other ST based targets - https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_NUCLEO_F439ZI/PinNames.h#L216

I'd like to use this as part of my device health IoT project.

CC @ARMmbed/team-ublox @RobMeades 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

fahim-ublox, 2018-12-12:
U-BLOX Internal ticket ID : PGM_NO_MBD-166

screamerbg, 2019-03-28:
@fahim-ublox Any progress on this?

fahim-ublox, 2019-03-29:
Pull request was merged, anything missed ? 
[9166](https://github.com/ARMmbed/mbed-os/pull/9166/)

0xc0170, 2019-08-14:
> Pull request was merged, anything missed ?

Should be resolved, I'll close this issue.

closed 2019-08-14 by 0xc0170