ARMmbed/mbed-os #9041: U-Blox C030 targets are missing BUTTON1

screamerbg, 2018-12-11:
### Description

According to the board page (https://os.mbed.com/platforms/ublox-C030-U201/), there is user button available on the C030 U201 (and other variants), but the pin is not defined as "BUTTON1", "SW1" and "USER_BUTTON". Here's an example how it's defined for ODIN W2 - https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/TARGET_MTB_UBLOX_ODIN_W2/PinNames.h#L163

CC @ARMmbed/team-ublox @RobMeades 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

screamerbg, 2018-12-11:
Also, it might be good to define LED color aliases given that there is color LED onboard. Example: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/TARGET_MTB_UBLOX_ODIN_W2/PinNames.h#L151

fahim-ublox, 2018-12-12:
Internal JIRA reference : https://jira.u-blox.net/browse/PGM_NO_MBD-167

closed 2018-12-19 by ciarmcom