ARMmbed/mbed-os #9071: UBLOX_EVK_ODIN_W2 wifi connection breaks after a few minutes

tommikas, 2018-12-12:
### Description

I modified mbed-cloud-client-example (2.1.0) to send notifications every second and then built the application for UBLOX_EVK_ODIN_W2. I used the onboard wifi to connect to a hotspot right next to the device.  Client connects fine and keeps sending notifications for some minutes, but typically before 10 minutes client loses connection and thereafter DNS resolution consistently fails. After resetting the device the same repeats.

Target: UBLOX_EVK_ODIN_W2
Toolchain: GCC_ARM 6.3.1
Tools: mbed-cli 1.8.3
Mbed OS: bf6f2c3c6434a6de9eb9511feffa5948b3d1f20f

Steps to reproduce:
1. Apply the modifications included below to the PDMC example application.
2. Configure wifi ssid and password.
3. Run the application.

```
diff --git i/configs/wifi_odin_v4.json w/configs/wifi_odin_v4.json
index 5ae9859..ad8a6a8 100644
--- i/configs/wifi_odin_v4.json
+++ w/configs/wifi_odin_v4.json
@@ -15,7 +15,7 @@
 "update-client.storage-address" : "(1024*1024*64)",
 "update-client.storage-size" : "(1024*1024*2)",
 "update-client.storage-locations" : 1,
- "mbed-trace.enable": null,
+ "mbed-trace.enable": 1,
 "nsapi.default-wifi-security" : "WPA_WPA2",
 "nsapi.default-wifi-ssid" : "\"SSID\"",
 "nsapi.default-wifi-password" : "\"Password\""
diff --git i/main.cpp w/main.cpp
index 204280e..b948989 100644
--- i/main.cpp
+++ w/main.cpp
@@ -132,7 +132,7 @@ void main_application(void)
 // make sure the line buffering is on as non-trace builds do
 // not produce enough output to fill the buffer
 setlinebuf(stdout);
-#endif 
+#endif
 
 // Initialize trace-library first
 if (application_init_mbed_trace() != 0) {
@@ -220,10 +220,8 @@ void main_application(void)
 
 while (mbedClient.is_register_called()) {
 static int button_count = 0;
- mcc_platform_do_wait(100);
- if (mcc_platform_button_clicked()) {
- button_res->set_value(++button_count);
- }
+ mcc_platform_do_wait(1000);
+ button_res->set_value(++button_count);
 }
 
 // Client unregistered, exit program.
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2018-12-12:
cc @ARMmbed/team-ublox

AmmadRehmat, 2019-01-02:
Does this scenario appears when the mbed traces are disabled?

cmonr, 2019-01-03:
@tommikas ^^^

screamerbg, 2019-02-11:
Bump @tommikas

AmmadRehmat, 2019-04-05:
@screamerbg @0xc0170 please close this if its not relevant anymore

0xc0170, 2019-04-05:
cc @SeppoTakalo @tommikas

SeppoTakalo, 2019-04-08:
Not relevant to us.
From the given log file, I can only see Client errors, and I know that it does not handle network state events, also the socket error is not shown, therefore impossible to tell if the device was working correctly or not.

tommikas, 2019-04-10:
I carved out some time to quickly test this.

What seems to happen is that the device loses connection to the AP and isn't able to reconnect until it is reset. Tracing doesn't seem to affect it.

aqib-ublox, 2019-05-28:
@todor603 your description matches with https://github.com/ARMmbed/mbed-os/issues/9621 and it being resolved in https://github.com/ARMmbed/mbed-os/pull/10454

please see discussion at https://github.com/ARMmbed/mbed-os/issues/9621

aqib-ublox, 2019-05-28:
Summary of issue handled under https://github.com/ARMmbed/mbed-os/issues/9621

`Client lost connection with AP and immediately try re-connecting that cause assertion and avoid re-connection so need to reset device` 

WIFI status was not handled in firmware that added and solve problem

tommikas, 2019-05-28:
That particular problem caused an assert though. This didn't.

aqib-ublox, 2019-05-28:
@tommikas can u test with latest mbed-os master branch because i didn't saw this problem in recent testing with https://github.com/ARMmbed/pelion-enablement

aqib-ublox, 2019-05-30:
@tommikas ?

tommikas, 2019-05-31:
> i didn't see this problem in recent testing with https://github.com/ARMmbed/pelion-enablement

Did you see it before #10454 was merged?

> can u test with latest mbed-os master branch

mbed-cloud-client-example 3.0.0 with the changes mentioned in the original post
mbed OS revision 9cc1caa0312c3332afed06f7d9b34b5753c95988

Seemed to work better in that the device ran for nearly an hour without the original issue happening again, but then I got an assert:
```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: ok == osOK
Location: 0x8061F2D
File: ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp+1602
Error Value: 0x0
Current Thread: application_unnamed_thread  Id: 0x20007E10 Entry: 0x8065B51 StackSize: 0x1000 StackMem: 0x20018970 SP: 0x200198B8
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=UBLOX_EVK_ODIN_W2
-- MbedOS Error Info --
```

tommikas, 2019-08-14:
Just tested again and didn't see this behavior anymore, including the assertion.

closed 2019-08-14 by tommikas