ARMmbed/mbed-os #9075: K82F QSPI test fails intermittently

cmonr, 2018-12-12:
### Description

Observed over the past couple of days, weeks, we've noticed that every once in a while, a greentea-test will fail, with the only failure being `K82F iar_arm / K82F-IAR.tests-mbed_hal-qspi.qspi frequency setting test`.

Below is the shortlog of the error:
```
[1544644866.12][CONN][RXD] >>> Running case #3: 'qspi frequency setting test'... 
[1544644866.22][CONN][INF] found KV pair in stream: {{__testcase_start;qspi frequency setting test}}, queued... 
[1544644867.12][CONN][RXD] tx data: 8B F5 77 C1 83 AD EF F9 9B 85 67 31 93 3D FF 89 54 78 2C D0 04 28 DC 20 B4 B3 0F 51 70 18 76 6A 6E 82 06 9A 3E 52 D6 A9 A4 97 CD 3A 68 0A F4 BB AB A3 87 15 A6 ED 35 B8 F2 DA 10 1F 19 7C E1 C8 2F A5 0E C3 16 71 C0 CC 5D 22 46 B7 AE 0D AA C2 4E 4F E3 08 94 39 3F 60 91 7A 7E A7 7D EB D3 DE E5 E6 55 01 C9 1C 92 DB 44 CB 32 B1 FB 5C 1A 1E 34 D7 B0 C5 03 9D 5F 21 A8 58 6C BC 02 E4 0B EE F3 E9 E7 00 0C 05 F8 BE 84 D2 75 2D 62 7F 41 AC 07 4D F0 BD EA 59 4A 17 56 C7 33 23 F7 4C 24 2B 09 26 86 C6 6D F6 3B 6F B6 5A 79 8A 61 99 B5 95 B9 42 8F 9C A0 48 1B F1 50 DD 98 D1 EC 72 8E D4 13 25 BA 2E 49 43 40 37 9F A2 96 4B 74 14 CF FD FE FC 38 45 66 E8 2A 90 5B 81 AF D9 53 63 FA 8D 27 5E 73 DF 6B 30 8C BF 28 CA A1 9E 06 57 3C 36 D5 C4 29 47 83 B2 12 6A 6E 88 8B 67 78 1D 64 D6 
[1544644868.03][CONN][RXD] rx data: CB F5 77 C1 83 AD EF F9 9B 85 67 31 93 3D FF 89 54 78 2C D0 04 28 DC 20 B4 B3 0F 51 70 18 76 6A 6E 82 06 9A 3E 52 D6 A9 A4 97 CD 3A 68 0A F4 BB AB A3 87 15 A6 ED 35 B8 F2 DA 10 1F 19 7C E1 C8 2F A5 0E C3 16 71 C0 CC 5D 22 46 B7 AE 0D AA C2 4E 4F E3 08 94 39 3F 60 91 7A 7E A7 7D EB D3 DE E5 E6 55 01 C9 1C 92 DB 44 CB 32 B1 FB 5C 1A 1E 34 D7 B0 C5 03 9D 5F 21 A8 58 6C BC 02 E4 0B EE F3 E9 E7 00 0C 05 F8 BE 84 D2 75 2D 62 7F 41 AC 07 4D F0 BD EA 59 4A 17 56 C7 33 23 F7 4C 24 2B 09 26 86 C6 6D F6 3B 6F B6 5A 79 8A 61 99 B5 95 B9 42 8F 9C A0 48 1B F1 50 DD 98 D1 EC 72 8E D4 13 25 BA 2E 49 43 40 37 9F A2 96 4B 74 14 CF FD FE FC 38 45 66 E8 2A 90 5B 81 AF D9 53 63 FA 8D 27 5E 73 DF 6B 30 8C BF 28 CA A1 9E 06 57 3C 36 D5 C4 29 47 83 B2 12 6A 6E 88 8B 67 78 1D 64 D6 
[1544644868.03][CONN][RXD] erase/write/read time: 37939/608/31 [us] 
[1544644868.03][CONN][RXD] :213::FAIL: Expected 139 Was 203 
[1544644868.13][CONN][INF] found KV pair in stream: {{__testcase_finish;qspi frequency setting test;0;1}}, queued... 
[1544644868.23][CONN][RXD] >>> 'qspi frequency setting test': 0 passed, 1 failed with reason 'Assertion Failed'
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

cmonr, 2018-12-12:
FYI @ARMmbed/mbed-os-storage @ARMmbed/mbed-os-hal @ARMmbed/mbed-os-test-team

offirko, 2018-12-13:
In targets.json K82F is configured with: "device_has" "QSPI".
AFAIK, K82F does not have QSPI - so it should be removed

cc: @maciejbocianski

offirko, 2018-12-13:
I wasn't aware that K82F was added with qspi_api.c hal.
Regardless of this issue, now that QSPI hal is supported, we will turn K82F default BD to QSPIF.

0xc0170, 2018-12-18:
cc @ARMmbed/team-nxp

0xc0170, 2018-12-18:
This again failed here https://github.com/ARMmbed/mbed-os/pull/9126#issuecomment-448170819

maciejbocianski, 2019-01-03:
It was easy to reproduce this fail, running test 20 times in loop
With the fix #9229 test passes (tested running 50 times in loop)

geky, 2019-05-16:
@0xc0170 this looks resolved, close issue?

maciejbocianski, 2019-06-04:
@0xc0170 yes please close

0xc0170, 2019-06-04:
As soon as PR is merged, will close.

closed 2019-08-14 by 0xc0170