ARMmbed/mbed-os #9082: Build fails CI on master (missing symbol 'main')

deece, 2018-12-13:
### Description

Testing against master commit 1c201b4628e346c62e1fbfa7cbcf6aa4e164f0c5

The following commands which are used as part of fail to build (missing symbol 'main'):
```
MBED_GCC_ARM_PATH=/usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin python tools/make.py -t GCC_ARM -m K64F --source=. --build=BUILD/K64F/GCC_ARM -j0 -v
...
Link: mbed-os
Preproc: /usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-cpp -E -P ./targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/device/TOOLCHAIN_GCC_ARM/MK64FN1M0xxx12.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x0 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x30000 -DMBED_RAM1_START=0x1fff0000 -DMBED_RAM1_SIZE=0x10000 -DMBED_BOOT_STACK_SIZE=1024 -o BUILD/K64F/GCC_ARM/.link_script.ld
[DEBUG] Return: 0
Link: /usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc @BUILD/K64F/GCC_ARM/.link_options.txt
[DEBUG] Return: 1
[DEBUG] Errors: BUILD/K64F/GCC_ARM/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: In function `__wrap_main':
[DEBUG] Errors: /home/adsilva/src/mbed-os/./rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:93: undefined reference to `main'
[DEBUG] Errors: collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "tools/make.py", line 73, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "/home/adsilva/src/mbed-os/tools/build_api.py", line 559, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 645, in link_program
    self.link(elf, objects, libraries, lib_dirs, linker_script)
  File "/home/adsilva/src/mbed-os/tools/hooks.py", line 34, in wrapper
    res = function(t_self, *args, **kwargs)
  File "/home/adsilva/src/mbed-os/tools/toolchains/gcc.py", line 265, in link
    self.default_cmd(cmd)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 674, in default_cmd
    raise ToolException(_stderr)
ToolException: BUILD/K64F/GCC_ARM/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: In function `__wrap_main':
/home/adsilva/src/mbed-os/./rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:93: undefined reference to `main'
collect2: error: ld returned 1 exit status
```

and

```
MBED_GCC_ARM_PATH=/usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin python tools/make.py -t GCC_ARM -m DISCO_F401VC --source=. --build=BUILD/DISCO_F401VC/GCC_ARM -j0 -v
...
Link: mbed-os
Preproc: /usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-cpp -E -P ./targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xC/device/TOOLCHAIN_GCC_ARM/STM32F401XC.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x40000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x10000 -DMBED_BOOT_STACK_SIZE=1024 -o BUILD/DISCO_F401VC/GCC_ARM/.link_script.ld
[DEBUG] Return: 0
Link: /usr/local/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc @BUILD/DISCO_F401VC/GCC_ARM/.link_options.txt
[DEBUG] Return: 1
[DEBUG] Errors: BUILD/DISCO_F401VC/GCC_ARM/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: In function `__wrap_main':
[DEBUG] Errors: /home/adsilva/src/mbed-os/./rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:93: undefined reference to `main'
[DEBUG] Errors: collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "tools/make.py", line 73, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "/home/adsilva/src/mbed-os/tools/build_api.py", line 559, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 645, in link_program
    self.link(elf, objects, libraries, lib_dirs, linker_script)
  File "/home/adsilva/src/mbed-os/tools/hooks.py", line 34, in wrapper
    res = function(t_self, *args, **kwargs)
  File "/home/adsilva/src/mbed-os/tools/toolchains/gcc.py", line 265, in link
    self.default_cmd(cmd)
  File "/home/adsilva/src/mbed-os/tools/toolchains/__init__.py", line 674, in default_cmd
    raise ToolException(_stderr)
ToolException: BUILD/DISCO_F401VC/GCC_ARM/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.o: In function `__wrap_main':
/home/adsilva/src/mbed-os/./rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:93: undefined reference to `main'
collect2: error: ld returned 1 exit status

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

deepikabhavnani, 2018-12-13:
Error indicates `main` is missing.  I believe command used here was 'mbed compile -m <target> -t <toolchain>`

main.cpp with main() function is needed for all the applications. See `https://github.com/ARMmbed/mbed-os-example-blinky` example (and https://os.mbed.com/teams/mbed-os-examples/code/) for reference. 

Also in order to compile tests provided in mbed-os repo (each tests is treated as standalone application and has main.cpp) command is
mbed test --compile -m <target> -t <toolchain>

deepikabhavnani, 2018-12-13:
Ohh I missed, you are using CI python scripts to build. `tools/make.py` is internal script not intended for user. If you want to use that follow the other tedious steps in CI as well, where main.cpp is generate from some Readme or code snippet at some place.

deece, 2018-12-13:
Ok, builds fine once there is a main.cpp, thanks.

closed 2018-12-13 by deece