ARMmbed/mbed-os #9083: FDS module in Nordic devices

deepikabhavnani, 2018-12-13:
### Description

https://github.com/ARMmbed/mbed-os/blob/f3424da060734fa9e27bf6a798747fb48ad63938/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/config/sdk_config.h#L3413

Flash data storage module of Nordic devices is enabled by default and adds to the code size of ~2K. 
Mbed OS has its own filesystem / block device support, which does not use this interface.
Functions in `TARGET_NORDIC\TARGET_NRF5x\TARGET_SDK_14_2\libraries\fds` are used by BLE module. Can we conditionally enable this dependency only when BLE is used, else move the dependent code to `C:\Users\deebha02\work_mbed\bootloader\mbed-bootloader-internal\mbed-os\features\FEATURE_BLE\targets\TARGET_NORDIC` ?

### Issue request type
    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2018-12-13:
@ARMmbed/mbed-os-pan Please review

desmond-blue, 2019-07-31:
@deepikabhavnani, we are now using Cordio as BLE module, it seems FDS no more needed even enabling BLE, perhaps we can just disable FDS in sdk_config.h?

deepikabhavnani, 2019-07-31:
Yes disabling seems good option. Nordic old sdk had lots of components enabled (resulted in flash usage) even if not required which might not be the case with new sdk.

closed 2019-10-31 by ciarmcom