ARMmbed/mbed-os #9093: nrf52: Periodic wakeups every 3ms

bearsh, 2018-12-13:
### Description

While using the ble heartrate example I looked at the current consumption and it really look like tickless mode isn't active. The cpu seems to wake up every 3ms which results in a quite high current consumption.

[image]

[image]

- nrf52840
- gcc
- 2fd0c5cfb (HEAD, tag: mbed-os-5.10.4, tag: latest, origin/mbed-os-5.10) Merge pull request #8775 from ARMmbed/release-candidate

### Issue request type

- [x] Question
- [ ] Enhancement
- [x] Bug

bearsh, 2018-12-21:
ok maybe this is related to the softdevice, I checked a example from the nordic sdk (v15), there we have a similar picture but waking up **only every 8.5ms**
[image]

cmonr, 2018-12-27:
@bearsh It would be interesting if you could look into measurements with this PR https://github.com/ARMmbed/mbed-os/pull/9019

@ARMmbed/mbed-os-pan Please take a look.

bearsh, 2018-12-27:
@cmonr will do so once I'm back at work... don't expect any results before 7th of January.

bearsh, 2019-01-07:
ok, I've done the measurements and it seems to be worse than before, no it wakes up **every 1.53ms**.

I wondering how the wakeup in the softdevice can be adjusted...

TacoGrandeTX, 2019-01-09:
@bearsh We've brought up a nRF52840_DK system using the Nordic Power Profiler Kit and have made some measurements using the Heart Rate Monitor Example (on v5.10.4 with the v14.2 SDK).  We aren't seeing the same periodic pattern as you are when the processor is idle.   

To get the best baseline numbers possible, we've removed the printf() and LED toggling and attached are waveforms when advertising.  We've been experimenting with the program and in this particular run we reduced the periodicCallback period to 1 second and the advertising interval to 2 seconds. When zooming in, the average program idle current is about 17uA.  It's a shame that the Nordic program doesn't autozoom the Y axis but the max is 29uA.  Based on the magnitude of the spikes that you are observing, I don't think we're experiencing them.  

How are you making your current measurement? 

Are you using a custom board?  

[image]
[image]

bearsh, 2019-01-10:
thanks @TacoGrandeTX for looking into this.

I've redone the measurements but getting the same results. My board is the ublox NINA-B3 Evk, the current is measured with a Agilent 34410A. I will also redo the measurements with our prototype board (once ready) with a NINA-B3 module.

Looking at your screenshots, the consumption still seems to high compared to what Nordic estimates in their [Online Power Profiler](https://devzone.nordicsemi.com/power/).

By the way, did you also look at the current once connected to and disconnected from a central? (like in #9276)

pan-, 2019-01-10:
@bearsh Measurement from @TacoGrandeTX look better than the one I estimated in the online profiler: 

[image]

DC/DC is not enabled by default as depending on the actual hardware it may or may not be available. It help for the power consumption; you can try it by calling `sd_power_dcdc_mode_set` when ble has been initialised.

bearsh, 2019-01-10:
@pan-, I've never worked with the hw power profiler (not the online version) but for me it seems the average current of about 18uA in TacoGrandeTX's screenshot is referenced only to the highlighter area and not to the whole x axis. But I glad to be proven wrong...

I know about the DC/DC converter and this seems to work just fine.

TacoGrandeTX, 2019-01-11:
I've run the HRM example on the EVK-NINA-B1 (modified for current measurement), but I only have a low-cost (and low sampling) DMM so there isn't much to learn by analyzing it.  When advertising the DMM steadies at 20 uA.  When connected it bounces around too much get a valid reading.  Note that this board is based on the nRF52832 so this isn't a good comparison to start with.  I don't think we can put much into this measurement. 

We did discover something interesting though.  The nRF52840 has a piece of IP called the CryptoCell310 (CC310) that is used to facilitate encryption.  By default that is enabled and used by Mbed.  However it appears to have high current demands as the floor goes above 5mA.  Have you observed the same?  If we revert back to a software routine for encryption (as on the nRF52832) then the floor goes back to a level similar to advertising.  Attached is a capture of the connection process. 

To prevent the CC310 from being used you can use this for your mbed_app.json:
```
    "target_overrides": {
        "NRF52840_DK": {
            "target.features_add": ["BLE"],
            "target.features_remove": ["CRYPTOCELL310"],
            "target.macros_remove": ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.macros_add": ["NRFX_RNG_ENABLED=1", "RNG_ENABLED=1", "NRF_QUEUE_ENABLED=1"]
           }
        }
```
Going back to your concern on the 3msec spikes, have you made any modifications to the BLE settings for your app?  

[image]

bearsh, 2019-01-12:
Thanks @TacoGrandeTX for looking into this.
I'm gonna test the 'crypto workaround' on Monday.

Regarding the spikes, the only modifications I've made was board related and I tried different advertising intervals.
In the meantime I also did some measurements on a PCA10040 which I saw laying around in the office. There I also saw spikes but with lower peaks.
By now I also have a prototype board for the product I work on (NINA-B3 based), there I also see spike but also with lower peaks.

Maybe the spikes are produced by a disturbing source in my office but it's really strange to see a change in the period of them based on used software.

I suggest we put this issue on hold as the energy of the spikes is rather low but concentrate on #9276. If I have some time, I will do more measurements on other nordic boards (PCA10059, NINA-B1, PCA10056) once I receive them.

TacoGrandeTX, 2019-02-13:
@bearsh Do you have any further observations of the periodic pulses?  As you are also seeing them on Nordic SDK15 example, have you consulted with them about it?

yogeshk19, 2019-03-10:
@TacoGrandeTX Do we still need to add the changes below in the mbed_app.json to reduce the current draw if we are not planning to use Cryptocell310? 

```
"target_overrides": {
        "NRF52840_DK": {
            "target.features_add": ["BLE"],
            "target.features_remove": ["CRYPTOCELL310"],
            "target.macros_remove": ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.macros_add": ["NRFX_RNG_ENABLED=1", "RNG_ENABLED=1", "NRF_QUEUE_ENABLED=1"]
           }
        }
```
Also is it good enough to just add the macro below to remove CryptoCell310 or do we need to add the other macros listed above? The reason I ask is that when I just add the macro below, my code doesn't compile.

"target.features_remove": ["CRYPTOCELL310"],

TacoGrandeTX, 2019-03-11:
@yogeshk19 You still need to use the ```target.macros_remove``` and ```target.macros_add``` lines.

desmond-blue, 2019-07-31:
@bearsh It seems this issue already addressed, do you need further help? Thanks.

desmond-blue, 2019-08-26:
@0xc0170 I guess we can close it since no feedback from reporter for almost one month?

40Grit, 2019-08-26:
A lot has changed since issue was opened.

* Cryptocell properly implemented 
* Softdevice support dropped

I'd close it.

closed 2019-08-26 by 0xc0170