ARMmbed/mbed-os #9095: Mbed HAL Sleep tests failing on FUTURE_SEQUANA_PSA

MarceloSalazar, 2018-12-13:
### Description

Mbed HAL Sleep tests failing on FUTURE_SEQUANA_PSA

Using Mbed OS 5.11-RC3

Test case: mbed-os-tests-mbed_hal-sleep

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

cc @lrusinowicz @alzix @orenc17 
cc @adbridge to add to known issue list for 5.11

orenc17, 2018-12-17:
the following test are failing:
tests-mbed-crypto-sanity - HF
tests-mbed_drivers-flashiap - HF
tests-mbed_drivers-rtc - HF
tests-mbed_hal-flash - HF
tests-mbed_hal-rtc - HF
tests-mbed_hal-rtc_reset - HF
tests-mbed_hal-rtc_time - HF
tests-mbed_hal-sleep - FAIL
tests-mbed_hal-sleep_manager - FAIL
tests-mbed_hal-trng - HF
tests-mbedtls-selftest - HF

* HF - Test crashes on HardFault
* FAIL - Test fails but doesn't crash

MarceloSalazar, 2019-03-14:
@orenc17 can you confirm whether this issue is considered resolved?

orenc17, 2019-03-14:
it isnt

MarceloSalazar, 2019-03-14:
@lrusinowicz can you please have a look and re-test?

lrusinowicz, 2019-05-13:
@MarceloSalazar 
I've rerun SEQUANA_PSA tests on latest (5.12.3) Mbed OS - they are passing, except of some devicekey testcases which are failing. But I believe this is a different issue (and, frankly, have no idea what it is).
[tests_sequana_psa_gcc_20190513.log](https://github.com/ARMmbed/mbed-os/files/3172521/tests_sequana_psa_gcc_20190513.log)

0xc0170, 2019-05-13:
Can you rerun them on master? there was at lesat one fix to device key (I recall one was eliminating some debug messages).

MarceloSalazar, 2019-06-26:
Closing ticket, as FUTURE_SEQUANA_PSA target is no longer supported.
FUTURE_SEQUANA continues to work as expected.

closed 2019-06-26 by MarceloSalazar