ARMmbed/mbed-os #9103: Runtime error with large arrays

masoudr, 2018-12-14:
### Description
I'm using `NanoPB` in my project. It defines some large arrays (about 1024 bytes) in the code. When I increase the string sizes or define `char` arrays with a size of 1024 or larger, the code stuck on runtime or throw [this exception](https://armmbed.github.io/mbedos-error/?error=0x80FF013D). On the other hand, the compiler output log shows only about 10% usage of data and ram.
I was wondering if there is an option to increase max buffer size for data or ram?
I also should mention that the code works fine on other custom based platforms like `platformio`.
My target is based on DISCO_F407VG. 
Thanks.
### Issue request type
    [ ] Question
    [ ] Enhancement
    [* ] Bug

kjbracey-arm, 2018-12-14:
The array is presumably being placed onto the stack of whichever thread the routine is running on. You will need to increase the stack size for thread(s) using that code, or consider changing the code to make the array static.

If this is just a general workspace buffer you've placed in `main`, so its lifetime is the whole program anyway, then making it static has no real drawback and will make it show up properly in the memory stats.

kjbracey-arm, 2018-12-14:
The config options `rtos.main-thread-stack-size` controls the stack size for main. It appears to be set to 1K by default for most platforms.