ARMmbed/mbed-os #9115: Exporting blinky example results in error

fluidblue, 2018-12-17:
### Description

When trying to export the blinky example the following error occurs:

	Traceback (most recent call last):
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 423, in <module>
	    main()
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 415, in main
	    ignore=options.ignore
	  File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 145, in export
	    notify.info("Using targets from %s" % targets_json)
	AttributeError: 'NoneType' object has no attribute 'info'

Steps to reproduce:
1. Use mbed-cli v1.8.3 to import Mbed OS 2 blinky example:
`mbed import https://mbed.org/teams/mbed/code/mbed_blinky/`
Hint: For my board (LPC11U68), only Mbed OS 2 is supported.
2. Export using
`mbed export -i mcuxpresso -m LPC11U68`

Solution:
Add the following to function `export` in `tools/project.py` (line 133):

    if not notify:
        notify=TerminalNotifier()

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

fluidblue, 2018-12-17:
After fixing with the proposed solution, another error occurs when running `mbed export -i mcuxpresso -m LPC11U68`:

    Traceback (most recent call last):
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 427, in <module>
        main()
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 419, in main
        ignore=options.ignore
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/project.py", line 195, in export
        ignore=ignore
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/export/__init__.py", line 285, in export_project
        macros=macros)
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/export/__init__.py", line 161, in generate_project_files
        exporter.generate()
      File "/Users/user/Downloads/ARM-mbed/mbed-cli_test2/mbed_blinky/.temp/tools/export/mcuxpresso/__init__.py", line 85, in generate
        self.libraries = libraries
    AttributeError: can't set attribute

A quick and dirty fix for this is the following:

- In file `tools/export/mcuxpresso/__init__.py`:
- Do not modify or set `self.libraries`, but instead copy the contents to e.g. `self.libraries_mcuxpresso` and use this new variable throughout the `mcuxpresso/__init__.py` file.

It seems to work, but there there may be better alternatives for fixing this.

0xc0170, 2018-12-17:
cc @ARMmbed/team-nxp

naveenkaje, 2019-02-15:
@fluidblue Thanks for submitting this issue. I have uploaded a fix here https://github.com/ARMmbed/mbed-os/pull/9742

TacoGrandeTX, 2019-03-29:
@fluidblue This is merged... so closing.  Please update the issue if you continue to have problems.

fluidblue, 2019-04-16:
@TacoGrandeTX 
@naveenkaje 
Unfortunately, the error persists.

I tried again using mbed-cli v1.10.0 (on macOS):

    $ mbed import https://mbed.org/teams/mbed/code/mbed_blinky/

    [mbed] Working path "/Users/user/Desktop/test" (directory)
    [mbed] Importing program "mbed_blinky" from "https://mbed.org/teams/mbed/code/mbed_blinky" at latest revision in the current branch
    [mbed] Adding library "mbed" from "https://os.mbed.com/users/mbed_official/code/mbed/builds" at rev #3a7713b1edbc
    [mbed] Unpacking library build "3a7713b1edbc" in "/Users/user/Desktop/test/mbed_blinky/mbed"
    [mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

    $ cd mbed_blinky/

    $ mbed export -i mcuxpresso -m LPC11U68

    [mbed] Working path "/Users/user/Desktop/test/mbed_blinky" (program)
    Scan: mbed_blinky
    Traceback (most recent call last):
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/project.py", line 423, in <module>
        main()
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/project.py", line 415, in main
        ignore=options.ignore
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/project.py", line 191, in export
        ignore=ignore
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/export/__init__.py", line 285, in export_project
        macros=macros)
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/export/__init__.py", line 161, in generate_project_files
        exporter.generate()
      File "/Users/user/Desktop/test/mbed_blinky/.temp/tools/export/mcuxpresso/__init__.py", line 85, in generate
        self.libraries = libraries
    AttributeError: can't set attribute
    [mbed] ERROR: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python" returned error.
           Code: 1
           Path: "/Users/user/Desktop/test/mbed_blinky"
           Command: "/Applications/MBEDCLI.app/Contents/Resources/miniconda/bin/python -u /Users/user/Desktop/test/mbed_blinky/.temp/tools/project.py -i mcuxpresso -m LPC11U68 --profile debug --source ."
           Tip: You could retry the last command with "-v" flag for verbose output
    ---

fluidblue, 2019-04-16:
@TacoGrandeTX 
I cannot reopen this issue, can you do this?

0xc0170, 2019-08-14:
@fluidblue Have you resolved your issue or this is still valid on lastest master?

fluidblue, 2019-08-14:
I did not resolve it as merge request (only as suggestion in this thread), therefore I assume the issue is still open.

closed 2021-10-20 by ciarmcom