ARMmbed/mbed-os #9116: Mbed Online IDE can not update mbed-os 2 library to latest version.

mertkapar, 2018-12-17:
### Description
Mbed Online IDE can not update mbed-os 2 library to latest version.
This problem is not valid for mbed-os 5.x

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

TacoGrandeTX, 2019-01-03:
@mertkapar We do see a problem with the "Update All" feature at the project level and will get this investigated.  If you do an "Update" with the Mbed library selected in the workspace, that operation does work so there is still a means to get the library updated.  Hopefully you've discovered this by now.

Thank you for reporting this.

cmonr, 2019-01-03:
@adbridge When/what was the last Mbed 2 release?

adbridge, 2019-01-04:
v164 8 weeks ago, we are due a new release shortly. This looks like either a tools or references issue.
@theotherjimmy @th3james

TacoGrandeTX, 2019-01-31:
@mertkapar Are you able to use the work around we describe above?  If so, we would like to close this as it will become redundant with forthcoming changes to the UI.

closed 2019-03-11 by TacoGrandeTX