ARMmbed/mbed-os #9117: Official support for `STM32F417` targets

embeddedteam103, 2018-12-17:
### Description

Are there any plans for official support for `F417` targets? The differences from `F407` are minimal at best. It seems the only change in mbed (other then mapping these targets) would be updating the exporter mapping of device names.
We are willing to contribute the mapping of an evaluation board we use internally (I need to find it's name first :) - it is almost identical to the DISCO F407 board in terms of mapping.
Are there any such plans?

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

jeromecoutant, 2018-12-17:
Hi

From ST point of view, there is no plan to add the support of STM32F417 in MBED.

But as you said, support of this chip should not be so difficult.
We tried to describe how to proceed in this wiki page:
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform

And specially with the python script tools\targets\STM32_gen_PeripheralPins.py
which will make for you the correct  PeripheralPins.c file

Finally we will also review your pull request.

jeromecoutant, 2019-01-29:
@0xc0170
Could set "devices: st" label
and close ? :-)

closed 2019-01-29 by 0xc0170