ARMmbed/mbed-os #9118: Support for targets with small internal flash

embeddedteam103, 2018-12-17:
### Description

We are using `mbed` on a target with very small internal flash. So small, we couldn't fit the `mbed` codebase with our (tiny) code of our own.
To resolve this we used a very ad-hoc approach: we changed the configuration of our IAR compiler - from using the full standard library to only partial.
This caused us to miss certain functionality of `mbed` - namely everything related to `FILE*` implementation in `mbed_retarget`.
To make the code compile, we just wrapped every place in `mbed` containing `FILE` with an `#ifdef` for our small target.
This is a terrible approach :) we want to avoid (as we might want to support `gcc` in the future which would require a complete overhaul of the approach).

If the use of this IAR flag (or similar ones) can be officially supported, then these `#ifdef`s should be part of `mbed` (we will give a proper PR for our fork).
What do you think is the right way to procede?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

embeddedteam103, 2019-04-29:
Explanation for closing?

linlingao, 2019-04-30:
Sorry, the issue was closed by mistake.

0xc0170, 2019-08-14:
We recently added minimal-printf (new profile). @embeddedteam103 do you have your work for us to review?

@kjbracey-arm @evedon this might interest you.

evedon, 2019-08-14:
It would be interesting indeed to review the PR as this would nicely complement the work we did on minimal-printf.

0xc0170, 2019-10-25:
@embeddedteam103 Can you update this ticket or is this for closure?

closed 2019-10-31 by ciarmcom